openmiko/openmiko #2: img uses hardware MAC adress instead of wyze one, also no hostname

NicolaiVdS, 2020-09-11:
so at the current state of this project the wifi adapter uses the hardware level MAC address instead of the one thats been assing by WYZE/brand, this is not realy a problem but since there is no host name assigned it can be hard to find it on your router

Proposed solution:

- add a host name eg. "openmiko-XXXX" where XXXX the last 4 chars of the mac to make it easy to see which cam is which
- if possible use the MAC address assigned by WYZE/brand since its printed on the back of the cam and can be found more easily on the routers page

NicolaiVdS, 2020-09-12:
after some digging the WYZE branded is stored in /params/config/.product_config``

tachang, 2021-09-21:
Right now this is fairly low priority so I am closing the issue. Also I assign openmiko now as the hostname. If you feel this is particularly important feel free to drop me a comment in Slack.

closed 2021-09-21 by tachang

DOSputin, 2022-01-08:
> Right now this is fairly low priority so I am closing the issue. Also I assign openmiko now as the hostname. If you feel this is particularly important feel free to drop me a comment in Slack.

This appears to be incorrect.  On most recent release, polling for hostname "openmiko" and it is not present. 

This presents considerable difficulty for those who wish to use open wifi networks with a MAC access control scheme. 

Can we re-open this and give it some attention?