openmiko/openmiko #7: [Q] Support for Wyze cam v2 HW revision

RohitJones, 2020-12-12:
Does openmiko support the new hardware revision of the wyze cam v2 ?
There is an [open issue]( https://github.com/EliasKotlyar/Xiaomi-Dafang-Hacks/issues/1616) at the Xiaomi-Dafang-Hacks repo for the same issue.

RohitJones, 2020-12-12:
To answer my own question, Yes. openmiko is compatible with the new hw revision (FCC ID: 2ANJHWYZEC02) of the waze cam v2.
This was tested using the v1.0.0 prebuild binary.
There was the issue of the cam not obtaining a DHCP lease thus unable to stream anything over the network.

tachang, 2020-12-13:
Hi. It should. FYI I've removed the v1.0.0 firmware because of bugginess. Please use the v0.0.5 one for now (or when I release a new one).

closed 2020-12-13 by tachang