openmiko/openmiko #9: Rotating feed

bacarp89, 2020-12-27:
Is there a way to do this? I've been digging through for a while now, and can't find any info, nor anything in the system that worked.

tachang, 2020-12-29:
Rotating the image you mean? Like a 90 degree rotation?

bacarp89, 2020-12-29:
Sorry for my lack of clarity, yes I'm trying to get it to 90 or 180 without
third party.

tachang, 2021-01-01:
What is consuming the feed? Can't you just rotate in VLC or rotate the camera physically?

charredchar, 2021-01-26:
I would like to know this as well. While a lot of software offers methods of rotation after receiving the stream, a lot also has issues. This is a common option for most security cameras as they get mounted as weird angles, being able to set this directly on the camera and not worry about it in other pieces of software is nice.

Case in point... OctoPrint has option, but every single thing in OctoPrint that uses the webcam needs to be set individually as well, it can be a hassle and not always play nice.

ndrwstn, 2021-03-02:
I'm in a similar situation and need to be able to rotate the image 180; I have one camera added to my NVR, but it has no mechanism to rotate the stream. So right now it's captured upside down.

charredchar, 2021-03-03:
> What is consuming the feed? Can't you just rotate in VLC or rotate the camera physically?

I could not find an option in VLC to do it, unless it's something you add to the URL, but this would be VLC only and no where else. You can't always rotate it physically, for example for me I have it mounted to a wall but facing sideways, along that wall. The feed is rotated 270 degrees CCW, which actually gives a perfect view of along the wall but rotated. This is why most security cameras include the option to rotate to a portrait view.

r2baka, 2021-03-03:
VLC can rotate the feed. Tools > Effects and filters
[image]

[image]

ndrwstn, 2021-03-04:
Unfortunately, I'm not consuming this in VLC; if the camera doesn't support the manipulation, I'm SOL. So right now I'm recording a camera upside down on my NVR.

My understanding isthmus firmware uses ffmpeg to encode the output, we should be able to pass a transpose parameter per [this article](https://ostechnix.com/how-to-rotate-videos-using-ffmpeg-from-commandline/) which would rotate the stream.

It would surprise me if this were already possible, I'm just not savvy enough to know how to do it.

tachang, 2021-03-04:
Alright I believe I will be able to implement this fairly easily. The hardware encoder itself supports it:

```
  // Flipping code
  // IMP_ISP_Tuning_SetISPVflip(IMPISP_TUNING_OPS_MODE_ENABLE);
  // IMP_ISP_Tuning_SetISPHflip(IMPISP_TUNING_OPS_MODE_ENABLE);
```
My thought is to put it settings.json:

```
{
  "comment": "3 encoders (1920x1080 H264, 1920x1080 MJPEG, and 640x80 H264",
  "flip_vertical": 0,
  "flip_horizontal": 0,
  ...
}

It seems like I can only flip for the whole camera (not individual sources).

The plants are awesome by the way. I didn't know you could grow them upside down :)

tachang, 2021-03-04:
This is in the latest version.

closed 2021-03-04 by tachang

charredchar, 2021-03-04:
> This is in the latest version.

Awesome, I can try this out next week. Just take the snippet you showed earlier for the settings.json and use 1 for which ever is like to flip?

tachang, 2021-03-04:
@charredchar Yup...let me know if it works or not!