openmiko/openmiko #12: Unable to connect after flash

jay3702, 2021-01-30:
I am using a new Wyze Cam V2. The flash operation succeeds, but it appears that the information in wpa_supplicant.conf is not being applied. When I replace the sdcard containing /config/overlay/etc/wpa_supplicant.conf, the yellow light continues to flash about once per second. After powering down and then examining the sdcard, I find two log files. The log files have some errors, but  also seem to indicate that video is being captured. 

This is in /var/log/autonight.log:  0curl: (7) Failed to connect to 127.0.0.1 port 8081: Connection refused

This is at the top of /var/log/messages: 
Jan  1 00:00:20 openmiko kern.notice kernel: [    0.000000] Linux version 3.10.14 (root@a9d303927ee3) (gcc version 4.7.4 (Buildroot 2016.02) ) #1 PREEMPT Fri Jan 15 20:43:03 UTC 2021
Jan  1 00:00:20 openmiko kern.info kernel: [    0.000000] bootconsole [early0] enabled
Jan  1 00:00:20 openmiko kern.warn kernel: [    0.000000] CPU0 RESET ERROR PC:74904C54
Jan  1 00:00:20 openmiko kern.info kernel: [    0.000000] CPU0 revision is: 00d00101 (Ingenic Xburst)
Jan  1 00:00:20 openmiko kern.info kernel: [    0.000000] FPU revision is: 00b70000

Near the end of the file are a number of lines like this:
Jan 30 19:24:50 openmiko local1.notice videocapture[293]: 1917494512: Current FPS: 12.50 / Channel 0
Jan 30 19:24:52 openmiko local1.notice videocapture[293]: 1915397360: Current FPS: 12.50 / Channel 1
Jan 30 19:24:54 openmiko local1.notice videocapture[293]: 1913300208: Current FPS: 12.50 / Channel 2

I tried resetting the configuration by holding the reset key for a few seconds. I observed blue flashes after a few seconds. The camera returns to slow yellow flashes. 

I reformatted the sdcard using "sd card formatter" (32GB card), and tried the wpa_supplicant again. I am using the contents of your example file, replacing only the network name and password. I could not find a physical wpa_supplicant.conf file on the site nor in the source code, so I created it manually using Visual Studio Code. I checked for the existence of \r (carriage control) and found none, only \n line ends.

After reformatting and re-copying wpa_supplicant.conf, still no joy. Incidentally, I have also tried creating a 512MB partition on previous attempts to load the Xiaomi DaFang firmware (also unsuccessful), but found no difference. Either format works fine to flash the factory firmware.

tachang, 2021-02-18:
Double check your wpa_supplicant.conf doesn't have weird windows line endings.
Also check the IP (mac address changes as well).

tachang, 2021-03-02:
@jay3702 Were you able to flash OpenMiko?

tachang, 2021-03-02:
I would try with the latest version since I try to replace line endings for you automatically in wpa_supplicant.conf.

icemanjeff, 2021-03-10:
Also having the same issue

tachang, 2021-03-10:
I believe the issue here was that the wpa_supplicant.conf file was actually wpa_supplicant.conf.txt

Removing the .txt appended at the end solved the problem.

Closing the issue.

icemanjeff, 2021-03-10:
How do I join the slack? Having and issue flashing after successfully flashing 2 other V2 cams.