openmiko/openmiko #13: Does not connect with wifi after flashsing

0xygen8, 2021-01-31:
After following the steps outlined I now have state not described in your guide

according to the seller : Xiaomi Mijia Xiaofang Dafang Smart Camera 1S 1080P New Version T20L Chip (looks like wyze v2)

1. try camera with stock firmware (OK)
2. download the 0.13 firmware renamed to demo.bin (placed on a 64G FAT formatted SDXC)
3. insert card into camera, pressed button wait for the blue led
4. after 40-50 seconds the light turns yellow 
5. does not connect to wifi
6. retried with /config/overlay/etc/openmiko.conf setting WIFI_MODULE=8189fs and WIFI_MODULE=8189es
7. opened the camera and looked on the wifi chip 8189ft

is there a way i can debug this further?

0xygen8, 2021-01-31:
setup_button set long!!!!!!!!!!!!!!!!!!!
Interface:  MMC
  Device 0: Vendor: Man 00001b Snr b75df701 Rev: 10.1 Prod: 00000.
            Type: Removable Hard Disk
            Capacity: 61119.0 MB = 59.6 GB (125171712 x 512)
Filesystem: FAT32 "OPENMIKO   "
the manufacturer 1c
SF: Detected FM25Q64

reading demo.bin
reading demo.bin
jiabo_au_check_cksum_valid!!!!!!!!!!!!!!!!!!!!!!!!
jiabo_idx=4
misc_init_r before change the blue_gpio
gpio_request lable = blue_gpio gpio = 39
misc_init_r after gpio_request the blue_gpio ret is 39
misc_init_r after change the blue_gpio ret is 0
jiabo_start=40000,jiabo_len=a90000
flash erase...
flash write...
misc_init_r after change the blue_gpio ret is 1
the manufacturer 1c
SF: Detected FM25Q64

--->probe spend 4 ms
SF: 2621440 bytes @ 0x40000 Read: OK
--->read spend 339 ms
## Booting kernel from Legacy Image at 80600000 ...
   Image Name:   Linux-3.10.14
   Image Type:   MIPS Linux Kernel Image (lzma compressed)
   Data Size:    2093432 Bytes = 2 MiB
   Load Address: 80010000
   Entry Point:  803d0730
   Verifying Checksum ... OK
   Uncompressing Kernel Image ... OK

Starting kernel ...

[    0.000000] Initializing cgroup subsys cpu
[    0.000000] Initializing cgroup subsys cpuacct
[    0.000000] Linux version 3.10.14 (root@a9d303927ee3) (gcc version 4.7.4 (Buildroot 2016.02) ) #1 PREEMPT Fri Jan 15 20:43:03 UTC 2021
[    0.000000] bootconsole [early0] enabled
[    0.000000] CPU0 RESET ERROR PC:6EC02405
[    0.000000] CPU0 revision is: 00d00101 (Ingenic Xburst)
[    0.000000] FPU revision is: 00b70000
[    0.000000] CCLK:712MHz L2CLK:356Mhz H0CLK:200MHz H2CLK:200Mhz PCLK:100Mhz
[    0.000000] Determined physical RAM map:
[    0.000000]  memory: 004f5000 @ 00010000 (usable)
[    0.000000]  memory: 0007b000 @ 00505000 (usable after init)
[    0.000000] User-defined physical RAM map:
[    0.000000]  memory: 06000000 @ 00000000 (usable)
[    0.000000] Initrd not found or empty - disabling initrd
[    0.000000] Zone ranges:
[    0.000000]   Normal   [mem 0x00000000-0x05ffffff]
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x00000000-0x05ffffff]
[    0.000000] Primary instruction cache 32kB, 8-way, VIPT, linesize 32 bytes.
[    0.000000] Primary data cache 32kB, 8-way, VIPT, no aliases, linesize 32 bytes
[    0.000000] pls check processor_id[0x00d00101],sc_jz not support!
[    0.000000] MIPS secondary cache 128kB, 8-way, linesize 32 bytes.
[    0.000000] Built 1 zonelists in Zone order, mobility grouping off.  Total pages: 24384
[    0.000000] Kernel command line: console=tty0 console=ttyS1,115200n8 mem=96M@0x0 ispmem=8M@0x6000000 rmem=24M@0x6800000 init=/linuxrc rootfstype=squashfs root=/dev/mtdblock2 rw mtdparts=jz_sfc:256k(boot),2048k(kernel),13504k(root),256k(config),256k(para),-(flag)
[    0.000000] PID hash table entries: 512 (order: -1, 2048 bytes)
[    0.000000] Dentry cache hash table entries: 16384 (order: 4, 65536 bytes)
[    0.000000] Inode-cache hash table entries: 8192 (order: 3, 32768 bytes)
[    0.000000] Memory: 91360k/98304k available (3877k kernel code, 6944k reserved, 1197k data, 492k init, 0k highmem)
[    0.000000] SLUB: HWalign=32, Order=0-3, MinObjects=0, CPUs=1, Nodes=1
[    0.000000] Preemptible hierarchical RCU implementation.
[    0.000000] NR_IRQS:418
[    0.000000] clockevents_config_and_register success.
[    0.000028] Calibrating delay loop... 709.42 BogoMIPS (lpj=3547136)
[    0.057700] pid_max: default: 32768 minimum: 301
[    0.062390] CPU 0 Unable to handle kernel paging request at virtual address 00000004, epc == 800cfe54, ra == 8050c4e0
[    0.073162] Oops[#1]:
[    0.075452] CPU: 0 PID: 0 Comm: swapper Not tainted 3.10.14 #1
[    0.081364] task: 804d8270 ti: 804ce000 task.ti: 804ce000
[    0.086832] $ 0   : 00000000 10001c00 00000002 804dd9ac
[    0.092129] $ 4   : 84002c80 000080d0 00000060 00000001
[    0.097427] $ 8   : 80590000 6d203836 6d696e69 203a6d75
[    0.102724] $12   : 00000000 0000002a 00000000 00000048
[    0.108022] $16   : 84002c80 00000000 000080d0 8052ae18
[    0.113320] $20   : 80580000 83fc4c8c 8050c4e0 00000000
[    0.118617] $24   : 00000003 80014710                  
[    0.123914] $28   : 804ce000 804cfe88 00000000 8050c4e0
[    0.129212] Hi    : 00000000
[    0.132126] Lo    : 00000032
[    0.135054] epc   : 800cfe54 kmem_cache_alloc+0x40/0x1c4
[    0.140426]     Not tainted
[    0.143264] ra    : 8050c4e0 pidmap_init+0x6c/0xdc
[    0.148107] Status: 10001c03.KERNEL EXL IE 
[    0.152345] Cause : 40808008
[    0.155258] BadVA : 00000004
[    0.158174] PrId  : 00d00101 (Ingenic Xburst)
[    0.162586] Modules linked in:
[    0.165683] Process swapper (pid: 0, threadinfo=804ce000, task=804d8270, tls=00000000)
[    0.173710] Stack : 00000000 00000000 00000000 803d1ec8 80590000 000002c5 804d0000 8052f154
.  804d0000 8052ae18 80580000 83fc4c8c 00000000 8050c4e0 804d0000 00008000
.  0000012d 80582e88 00000000 00000000 804d0000 805058d8 00000000 8052ae18
.  804cdbb8 00000043 ffffffff ffffffff 805052a4 00000000 8052f154 00000000
.  ffffffff 83f98000 83fcb894 81f73d8c 83f994c4 83f99524 00000000 00000000
.  ...
[    0.209734] Call Trace:
[    0.212213] [<800cfe54>] kmem_cache_alloc+0x40/0x1c4
[    0.217251] [<8050c4e0>] pidmap_init+0x6c/0xdc
[    0.221752] [<805058d8>] start_kernel+0x344/0x3d8
[    0.226510] 
[    0.228010] 
Code: 24420001  af820014  8e110000 <8e340004> 8f820014  2442ffff  af820014  8f820008  30420004 
[    0.238122] ---[ end trace 4eaa2a86a8e2da22 ]---
[    0.242762] Kernel panic - not syncing: Attempted to kill the idle task!
[    0.249556] Rebooting in 3 seconds..

bikemike, 2021-02-09:
I have a similar camera. I don't think it's a T20L though. It has 64mb ram. I do not get a flashing yellow LED with the demo.bin (v0.0.13).

With the bin file from dafang hacks(cfw-1.0.bin) the yellow LED does flash at the end of the procedure.

0xygen8, 2021-02-09:
Is it possible for me to switch to dafang at this moment?

bikemike, 2021-02-09:
Yes. It's here: https://github.com/EliasKotlyar/Xiaomi-Dafang-Hacks

0xygen8, 2021-02-09:
Thanks, I'll do that

tachang, 2021-02-18:
OpenMiko does not work with 64MB of ram.

bikemike, 2021-02-19:
> OpenMiko does not work with 64MB of ram.

The project description has "Xiaomi Xiaofang 1S" in it. Perhaps that should be taken out?

tachang, 2021-02-19:
I think some `Xiaomi Xiaofang 1S` models have 128MB of ram. If they do I will add a note on the readme that it only supports 128MB versions of them.

tachang, 2021-03-02:
Closing out this issue since OpenMiko does not support 64mb ram cameras right now.

closed 2021-03-02 by tachang