openmiko/openmiko #14: Adding micropython SSL support for upip

qrimp, 2021-02-03:
First: I LOVE this project. It's totally expanding the world that I can do this to these cameras. Thank you.

Second: I'm supernoob on this, but doing my best.

Next, a question/request. The big picture is, I'm trying to add OpenCV code to do image processing on the camera itself. Using upip to install modules fails however, like this:
```
# micropython testr.py
Traceback (most recent call last):
  File "testr.py", line 3, in <module>
  File "/usr/lib/micropython/urequests.py", line 108, in get
  File "/usr/lib/micropython/urequests.py", line 44, in request
ImportError: no module named 'ussl'
```

So then, I tried to install ussl, like this, which failed.
```
# micropython -m upip install MICROPY_PY_USSL
Traceback (most recent call last):
  File "upip.py", line 110, in <module>
ImportError: no module named 'ussl'
```

So I looked into it, and it appears USSL is not included in the make. Can we add that?
\custompackages\package\micropython\micropython.mk

From this similar issue: https://github.com/micropython/micropython-lib/issues/245 

I believe I traced it there, perhaps changing it to MICROPY_PY_USSL=1, or simply removing that line would add USSL to our micropython powers on the chip?

\custompackages\package\micropython\micropython.mk
```
# When building from a tarball we don't have some of the dependencies that are in
# the git repository as submodules
MICROPYTHON_MAKE_OPTS = MICROPY_PY_BTREE=0
MICROPYTHON_MAKE_OPTS += MICROPY_PY_USSL=0
```

I don't know what that means, but I think that's a clue. Maybe I'm barking up the wrong tree? Is OpenCV on micropython using these boards even possible?

Regardless, having SSL requests via micropython on these boards would be great, and currently not possible from what I can tell. As is, I don't think I can use upip to install packages, either, which limits our capabilities.

I love this project. I'm happy to contribute, but I'm also a github noob, so there's a learning curve for me to do that. I don't even know what a pull request is or how to do one.

tachang, 2021-02-18:
Hi @r2baka thanks for trying OpenMiko out.

> First: I LOVE this project. It's totally expanding the world that I can do this to these cameras. Thank you.
>
> Second: I'm supernoob on this, but doing my best.
>
> Next, a question/request. The big [...]

So micropython and upip isn't configured. I was mostly beta testing some code and some way to get python to work but never really got deep enough to troubleshoot the issues before I realized that it would be a ton of space and I couldn't fit things in the image.

> So I looked into it, and it appears USSL is not included in the make. Can we add that?
> \custompackages\package\micropython\micropython.mk
> 
> From this similar issue: [micropython/micropython-lib#245](https://github.com/micropython/micropython-lib/issues/245)
> 
> I believe I traced it there, perhaps changing it to MICROPY_PY_USSL=1, or simply removing that line would add USSL to our micropython powers on the chip?
> 
> \custompackages\package\micropython\micropython.mk
> 
> ```
> # When building from a tarball we don't have some of the dependencies that are in
> # the git repository as submodules
> MICROPYTHON_MAKE_OPTS = MICROPY_PY_BTREE=0
> MICROPYTHON_MAKE_OPTS += MICROPY_PY_USSL=0
> ```
> 
> I don't know what that means, but I think that's a clue. Maybe I'm barking up the wrong tree? Is OpenCV on micropython using these boards even possible?

I think you are on the right track. I'm not sure OpenCV is possible on these boards but I don't see why not (barring disk space constraints). It might be super slow.

tachang, 2021-09-21:
Not planning to add micropython ssl at this time. In fact micropython is probably too heavyweight at this point. I might reconsider in the future.

closed 2021-09-21 by tachang