openmiko/openmiko #15: Wyzecam v2 Fails to join Wifi

Nitesedge, 2021-02-04:
New WyzeCam v2 arrived from Amazon today.

Firmware loads fine and works (confirmed via logfile written to SD card), but fails to join Wifi

```
Jan  1 00:00:27 openmiko kern.warn kernel: [   27.064958] RTL871X: HW EFUSE
Jan  1 00:00:27 openmiko kern.warn kernel: [   27.068065] RTL871X: hal_com_config_channel_plan chplan:0x20
Jan  1 00:00:27 openmiko kern.warn kernel: [   27.172017] RTL871X: rtw_regsty_chk_target_tx_power_valid return _FALSE for band:0, path:0, rs:0, t:-1
Jan  1 00:00:27 openmiko kern.warn kernel: [   27.183443] RTL871X: rtw_ndev_init(wlan0) if1 mac_addr=b4:c9:b9:70:1b:c9
Jan  1 00:00:27 openmiko kern.warn kernel: [   27.218838] RTL871X: module init ret=0
Jan  1 00:00:29 openmiko kern.warn kernel: [   29.925070] RTL871X: nolinked power save enter
```

The RTL871x doesn't match the documented doesn't match the documented "WIFI_MODULE=8189fs". Has the NIC changed or is this a red herring.

Worth noting I tried various WIFI APs with the range of security schemas including an open AP. I have configured linux via wpa_supplicant in the past so unless I am missing something looking like maybe a hardware change in the V2???

$ cat wpa_supplicant.conf 
```
ctrl_interface=/var/run/wpa_supplicant
ctrl_interface_group=0
ap_scan=1
network={
    ssid="apname"
    key_mgmt=None
}
```

messages file renamed to messages.log and attached
[messages.log](https://github.com/openmiko/openmiko/files/5928729/messages.log)

qrimp, 2021-02-05:
can you try with a network block like this on a secure network and see if it works?

```
network={
        ssid="yourssid"
        psk="yourpassword"

        key_mgmt=WPA-PSK
        pairwise=CCMP TKIP
        group=CCMP TKIP WEP104 WEP40
        priority=2
}
```

wanderson9, 2021-02-06:
I've been seeing posts on Dafang and here that some people are having some issues with newer devices, but I just want to provide a datapoint for you that I got a Wyzev2 from Amazon that arrived Wednesday and I had no problem getting it working.

wanderson9, 2021-02-06:
I've been seeing posts on Dafang and here that some people are having some issues with newer devices, but I just want to provide a datapoint for you that I got a Wyzev2 from Amazon that arrived Wednesday and I had no problem getting it working.

EDIT: $ cat /config/overlay/etc/wpa_supplicant.conf

```
# This file is used to configure wifi
#
# For more configuration option please see:
# https://w1.fi/cgit/hostap/plain/wpa_supplicant/wpa_supplicant.conf

ctrl_interface=/var/run/wpa_supplicant
ctrl_interface_group=0
ap_scan=1

network={
        ssid="INSERT_SSID_HERE"
        # Uncomment to connect to Hidden SSIDs
        #scan_ssid=1 
        key_mgmt=WPA-PSK
        pairwise=CCMP TKIP
        group=CCMP TKIP WEP104 WEP40
	# psk stands for pre-shared key (the wifi password)
        psk="INSERT_SSID_PASSWORD_HERE"
        priority=2
}

network={
        ssid="INSERT_SSID2_HERE"
        #scan_ssid=1 
        key_mgmt=WPA-PSK
        pairwise=CCMP TKIP
        group=CCMP TKIP WEP104 WEP40
	# psk stands for pre-shared key (the wifi password)
        psk="INSERT_SSID2_PASSWORD_HERE"
        priority=3 
```

Nitesedge, 2021-02-09:
Thanks for that. Good to know the hardware hasn't changed.

I did finally get it working and for those that follow there are a few lessons learned:

1. Once the camera joins the wifi it continues to flash yellow. Since I was expecting it to change solid blue once connected (like it does with the original firmware) I may have actually had it working at one point and did not realize.
2. I setup a dedicated BGN router with WPA (not WPA2) and the default wpa_supplicant.conf worked fine.

closed 2021-02-09 by Nitesedge