openmiko/openmiko #16: RTSP not working

hillm5714, 2021-02-09:
I have installed the latest release (Release v0.0.13) onto my WyzeCam V2. The stream and snapshot on port 8080 work just fine, but the RTSP feeds on 8554 do not work for me. Is there something else I need to configure other than loading the firmware, and adding the configured wpa_supplicant.conf file? Also, which of these feeds work best with OctoPrint? Is there anything else I should configure for this to work well with OctoPrint? Thanks in advance.

qrimp, 2021-02-09:
I would suggest trying VLC. When I first got these cameras working I was trying to access the 8554 streams with Firefox and it kept thinking I was trying to search.
[image]

[image]

[image]

hillm5714, 2021-02-10:
That did the trick. Thanks for the help qrimp.

closed 2021-02-10 by hillm5714