openmiko/openmiko #17: Image (night mode?) colors

mrspirytus, 2021-02-13:
Yesterday I flushed version 0.0.13 on the new Wyze v2. After booting I was able to connect to Wyze with vlc and I was able to stream rtsp just fine. The image looked great during the day light. I left the camera running overnight but vlc was not streaming. Today middle of the day I connected with vlc again and I am getting an image bit it looks like color channels got corrupt or maybe night mode did not restore properly?

Here is what I did to flush Wyze
- Format sdcard fat32
- copy 0.0.13 bin image and renamed to demo.bin
- create /config/overlay/etc/wpa_supplicant.conf with template and edited SSID and password.

No other changes or configurations were added. Please see the screenshot of how it looked when I connected it to the camera today. After I powered off and on camera, colors returned just fine.

Please let me know if I need to provide any other information or what configurations I am missing :)

[image]

qrimp, 2021-02-13:
That's normal. The original software converts that image to grayscale before displaying it to you. This firmware does _not_ change the color scale. It just shows you what the sensor receives. If you look at the physical camera in nightmode through your cell phone camera, you'll notice the LEDs are a pinkish color.

[image]

mrspirytus, 2021-02-13:
@qrimp Not sure what phone camera might have to do with this.
Here is an image I see now after the camera reboot? using vlc. Both screenshots are taken running vlc on my desktop using screen capture software vs taking picture of my screen with the camera (if that is what you were trying to explain)
I am thinking that yesterday I left the camera running mid-day. Left the camera overnight with nothing connected to it until today. after I rebooted the camera image looks like yesterday. Is this some night mode that was switched on and never reverted back next day?

[image]

qrimp, 2021-02-13:
Gotcha! The answer to your last question is, Yes.

This is a known issue. Sometimes the cameras get stuck in nightmode. It's a known issue. When the camera is in nightmode with the new firmware, the image is pink. Sometimes it gets stuck in nightmode, so it'll be pink during the day too.

The cell phone camera can "see" the nightmode IR LEDs on the camera. It's just a way to see them. Has nothing to do with this issue. I just suggested that so you could see why the photo is pink. It's because the IR LEDs are pink. Your eyes can't see the pink light they emit, but the cell phone camera can. You can too if you look at them, but I don't recommend that.

If you look at the image the camera produces at night, it'll be pink too. The pink color is normal and expected, however, I think the issue here is that it didn't switch out of nightmode until you rebooted the camera, which is a known issue. I think on the original project this one was forked from.

My tomatoes look like this at night:
[image]

mrspirytus, 2021-02-13:
@qrimp Ok thank you for an explanation. I am clear on the phone camera you suggested. I was initially under impression that I could stream the image to phone vs vlc.

Do you know if the mentioned issue might be fixed "relatively" soon? Or at this point, no one knows why it is happening and it is under investigation?

qrimp, 2021-02-14:
No idea. It was only implemented a handful of weeks ago. These things are little computers that run Linux. Frankly, I'm blown away by the potential of what's going on here. I'm still wrapping my head around it. It's not easy to do these things and autonightmode is a frustration for me too, but I have no expectations. 

I'm still in awe that this is even possible.

mrspirytus, 2021-02-15:
Fair enough :) I was not asking. Is there an option to disable night mode for now till this is resolved? So I do not have to reboot each day?

qrimp, 2021-02-15:
* gracefully bows out *

tachang, 2021-02-18:
> Fair enough :) I was not asking. Is there an option to disable night mode for now till this is resolved? So I do not have to reboot each day?

Hi. You can edit `openmiko.conf` and put it on the overlay directory:

https://github.com/openmiko/openmiko/blob/master/overlay_minimal/etc/openmiko.conf#L16

tachang, 2021-02-19:
Also the purple tint is due to IR and is a known issue. That is why a lot of nightmode things are black and white. I haven't gotten my hands on a WyzeCam V3 yet but they might do some post processing to get it to have bright colors at night. Not sure.

mrspirytus, 2021-02-19:
@tachang Awsome. Thank you for your reply and help
I copy above to SD card  /config/overlay/openmiko.conf and replace `ENABLE_AUTONIGHT=0` to disable night mode for now. Do I have to hold the 3-sec button on the power-up for it to take effect?

tachang, 2021-02-19:
Make sure you do on the sdcard `config/overlay/etc/openmiko.conf`.After that a simple reboot without holding anything down should work (with sdcard inside)

mrspirytus, 2021-02-19:
@tachang While I am copying openmiko.conf to my cameras, I got a question about few additional settings in this file.
Since I am only interested about rtsp streaming and no night vision for now is it ok to disable other settings? Any line like 
```
# Use a swapfile on the sdcard
#ENABLE_SWAP=1
```
Does this show a default value, like, for example, sshd_config has? Or is it that swap is off and for that matter, all listed items are off" unless explicitly stated?

So for my rtsp case would that be ok?
```
# Starts ffmpeg on startup recording from the v4l2rtspserver
ENABLE_RECORDING=0

# Start the micropython based api
ENABLE_API=0

# Use a swapfile on the sdcard
ENABLE_SWAP=0

# Enable MJPEG over HTTP
ENABLE_MJPEG_OVER_HTTP=0

# Script that detects nighttime/daytime and turns on IR LEDs
ENABLE_AUTONIGHT=0

WIFI_MODULE=8189fs

```

tachang, 2021-02-19:
I think for now all values are explicitly off unless stated. I may have done some stuff where it is defaulted on like sshd_config style but it's been a while and I'd have to check. What you have looks correct.

closed 2021-04-22 by tachang