openmiko/openmiko #21: Support for the Wyze Cam v3?

satmandu, 2021-03-22:
Anything we can do to pull together information to help make this happen?

smith288, 2021-04-06:
I dont think the v3 is a Ingenic T20 based device so probably not.

satmandu, 2021-04-06:
It's apparently an Ingenic T31 based device as per https://www.reddit.com/r/wyzecam/comments/k0zocr/looks_like_the_v3_is_using_the_ingenic_t31_which/ 
There is some active work on getting uboot working on it as per https://gitter.im/Xiaomi-Dafang-Hacks/Lobby?at=6067b10d3153ce63a3b6d28d

tachang, 2021-09-21:
I'd love to support WyzeCam V3 if I can get my hands on one. Mostly sold out. Closing this out for now.

closed 2021-09-21 by tachang

beankylla, 2022-02-27:
@tachang  would also be interested in getting this. I think a LOT of people would be. If it's just down to getting you one, i think we could make that happen ! Are you still on?

hmmcclish, 2022-04-11:
Thirding the offer of a Wyze Cam v3 if you'd still have any interest in supporting it, @tachang - just got mine from Amazon this week!