openmiko/openmiko #23: Cannot seem to get wifi connected - dafang hacks connects but not openmiko.

bad1dea, 2021-05-04:
Having issues getting wifi connected, not seeing much in the logs.

created/copied file to sd card as per instructions (mSD - config  -> overlay -> etc -> wpa_supplicant.conf)

log files are being created, and things are running, just not connecting to wifi, do not see mac or additional mac on router (pfsense)

all my dafang-hacks cameras are connected fine, trying out openmiko but having connection issues obviously.. let me know if I'm doing something wrong?

<details>
<summary>wpa_supplicant.conf (obviously removed my ssid/pw)</summary>
<p>

```
# This file is used to configure wifi
#
# For more configuration option please see:
# https://w1.fi/cgit/hostap/plain/wpa_supplicant/wpa_supplicant.conf

ctrl_interface=/var/run/wpa_supplicant
ctrl_interface_group=0
ap_scan=1

network={
        ssid="#######"
        # Uncomment to connect to Hidden SSIDs
        #scan_ssid=1 
        key_mgmt=WPA-PSK
        pairwise=CCMP TKIP
        group=CCMP TKIP WEP104 WEP40
        # psk stands for pre-shared key (the wifi password)
        psk="#########"
        priority=2
}
```

\```

</p>
</details>  

<details>
<summary>messages log</summary>
<p>

```
Jan  1 00:00:23 openmiko syslog.info syslogd started: BusyBox v1.24.1
Jan  1 00:00:23 openmiko kern.notice kernel: klogd started: BusyBox v1.24.1 (2021-04-23 00:11:31 UTC)
Jan  1 00:00:23 openmiko kern.info kernel: [    0.000000] Initializing cgroup subsys cpu
Jan  1 00:00:23 openmiko kern.info kernel: [    0.000000] Initializing cgroup subsys cpuacct
Jan  1 00:00:23 openmiko kern.notice kernel: [    0.000000] Linux version 3.10.14 (root@7680ee476c6d) (gcc version 4.7.4 (Buildroot 2016.02) ) #1 PREEMPT Fri Apr 23 00:44:46 UTC 2021
Jan  1 00:00:23 openmiko kern.info kernel: [    0.000000] bootconsole [early0] enabled
Jan  1 00:00:23 openmiko kern.warn kernel: [    0.000000] CPU0 RESET ERROR PC:9E4A1398
Jan  1 00:00:23 openmiko kern.info kernel: [    0.000000] CPU0 revision is: 00d00101 (Ingenic Xburst)
Jan  1 00:00:23 openmiko kern.info kernel: [    0.000000] FPU revision is: 00b70000
Jan  1 00:00:23 openmiko kern.debug kernel: [    0.000000] cgu_get_rate, parent = 860160000, rate = 0, m = 0, n = 0, reg val = 0x000020ff
Jan  1 00:00:23 openmiko kern.warn kernel: [    0.000000] CCLK:860MHz L2CLK:430Mhz H0CLK:200MHz H2CLK:200Mhz PCLK:100Mhz
Jan  1 00:00:23 openmiko kern.info kernel: [    0.000000] Determined physical RAM map:
Jan  1 00:00:23 openmiko kern.info kernel: [    0.000000]  memory: 004f7000 @ 00010000 (usable)
Jan  1 00:00:23 openmiko kern.info kernel: [    0.000000]  memory: 00079000 @ 00507000 (usable after init)
Jan  1 00:00:23 openmiko kern.info kernel: [    0.000000] User-defined physical RAM map:
Jan  1 00:00:23 openmiko kern.info kernel: [    0.000000]  memory: 06000000 @ 00000000 (usable)
Jan  1 00:00:23 openmiko kern.info kernel: [    0.000000] Initrd not found or empty - disabling initrd
Jan  1 00:00:23 openmiko kern.warn kernel: [    0.000000] Zone ranges:
Jan  1 00:00:23 openmiko kern.warn kernel: [    0.000000]   Normal   [mem 0x00000000-0x05ffffff]
Jan  1 00:00:23 openmiko kern.warn kernel: [    0.000000] Movable zone start for each node
Jan  1 00:00:23 openmiko kern.warn kernel: [    0.000000] Early memory node ranges
Jan  1 00:00:23 openmiko kern.warn kernel: [    0.000000]   node   0: [mem 0x00000000-0x05ffffff]
Jan  1 00:00:23 openmiko kern.debug kernel: [    0.000000] On node 0 totalpages: 24576
Jan  1 00:00:23 openmiko kern.debug kernel: [    0.000000] free_area_init_node: node 0, pgdat 80503cf0, node_mem_map 81000000
Jan  1 00:00:23 openmiko kern.debug kernel: [    0.000000]   Normal zone: 192 pages used for memmap
[... 309 lines cut ...]
Jan  1 00:02:00 openmiko cron.info crond[114]: file root:
Jan  1 00:02:00 openmiko cron.info crond[114]:  line /usr/sbin/logrotate /etc/logrotate.conf
Jan  1 00:02:03 openmiko local1.notice videocapture[279]: 1917555952: Current FPS: 25.00 / Channel 2
Jan  1 00:02:05 openmiko local1.notice videocapture[279]: 1919653104: Current FPS: 25.00 / Channel 1
Jan  1 00:02:07 openmiko local1.notice videocapture[279]: 1921750256: Current FPS: 25.00 / Channel 0
Jan  1 00:02:11 openmiko local1.notice videocapture[279]: 1917555952: Current FPS: 25.00 / Channel 2
Jan  1 00:02:14 openmiko local1.notice videocapture[279]: 1919653104: Current FPS: 25.00 / Channel 1
Jan  1 00:02:15 openmiko local1.notice videocapture[279]: 1921750256: Current FPS: 25.00 / Channel 0
Jan  1 00:02:19 openmiko local1.notice videocapture[279]: 1917555952: Current FPS: 25.00 / Channel 2
Jan  1 00:02:22 openmiko local1.notice videocapture[279]: 1919653104: Current FPS: 25.00 / Channel 1
Jan  1 00:02:23 openmiko local1.notice videocapture[279]: 1921750256: Current FPS: 25.00 / Channel 0
Jan  1 00:02:27 openmiko local1.notice videocapture[279]: 1917555952: Current FPS: 25.00 / Channel 2
Jan  1 00:02:31 openmiko local1.notice videocapture[279]: 1919653104: Current FPS: 25.00 / Channel 1
Jan  1 00:02:32 openmiko local1.notice videocapture[279]: 1921750256: Current FPS: 25.00 / Channel 0
Jan  1 00:02:35 openmiko local1.notice videocapture[279]: 1917555952: Current FPS: 25.00 / Channel 2
Jan  1 00:02:39 openmiko local1.notice videocapture[279]: 1919653104: Current FPS: 25.00 / Channel 1
Jan  1 00:02:40 openmiko local1.notice videocapture[279]: 1921750256: Current FPS: 25.00 / Channel 0
Jan  1 00:02:44 openmiko local1.notice videocapture[279]: 1917555952: Current FPS: 25.00 / Channel 2
Jan  1 00:02:48 openmiko local1.notice videocapture[279]: 1921750256: Current FPS: 25.00 / Channel 0
Jan  1 00:02:48 openmiko local1.notice videocapture[279]: 1919653104: Current FPS: 25.00 / Channel 1

\```

</p>
</details>

bad1dea, 2021-05-04:
To add some notes..

I've gone through the troubleshooting
  - no windows line endings/new lines, (I always use unix style formatting)
  - these are original dafangs
  - filenames are correct (no additional .txt or any others)
  - I've tried multiple mSD cards all formatted to fat32 with same directory structure.

bad1dea, 2021-05-04:
So.. after reading the closed comments, looks like there was a setting in openmiko.conf that corrected it.. wrong driver was being loaded for the pan.

My suggestion would be to add under the setting up wifi to validate/create that file. Will be easier for future users.

closed 2021-05-04 by bad1dea

sorryusernameisalreadytaken, 2022-02-03:
> 

Can you explain your solution plese :)
What is the openmiko.conf amd what content should it have.
'''
Jan  1 00:00:26 openmiko user.notice root: Starting wifi using module 8189fs
Jan  1 00:00:26 openmiko kern.warn kernel: [   26.274608] RTL871X: module init start
Jan  1 00:00:26 openmiko kern.warn kernel: [   26.278506] RTL871X: rtl8189fs v4.3.24.8_22657.20170607
Jan  1 00:00:26 openmiko kern.warn kernel: [   26.288339] ######platform_wifi_power_on: 
Jan  1 00:00:26 openmiko kern.info kernel: [   26.340378] mmc1: new SDIO card at address 0001
Jan  1 00:00:27 openmiko kern.warn kernel: [   27.633456] RTL871X: module init ret=0
Jan  1 00:00:27 openmiko user.notice root: Using hostname: openmiko
Jan  1 00:00:27 openmiko user.notice root: Attempting DHCP
Jan  1 00:00:27 openmiko user.notice root: Starting ntpd: 
Jan  1 00:00:27 openmiko user.notice root: FAIL
Jan  1 00:00:28 openmiko authpriv.info dropbear[208]: Running in background
'''

bill-mcgonigle, 2023-04-25:
> @bad1dea Can you explain your solution plese :) What is the openmiko.conf amd what content should it have.

@sorryusernameisalreadytaken : probably this?

```
# grep -H 8189 /etc/openmiko.conf
/etc/openmiko.conf:# Use 8189fs for WyzeCam V2. For the WyzeCam Pan and Dafang use 8189es
/etc/openmiko.conf:WIFI_MODULE=8189fs

```