openmiko/openmiko #24: Unable to flash on 3 WYZE v2s

RobD83, 2021-05-07:
I have been trying for several days to get this up and running on 3 different v2s. One purchased around Aug-Oct 2020 and two in Feb 2021. All 3 result in the same outcome: nothing written to SD card and a yellow light bootloop situation. After "flashing" ir filter clicks, yellow light flashes for ~30 sec, ir filter click, off, repeat. I've gotten dafang operating on 2 of them with no problems, haven't tried the 3rd as I really want OM for the mjpeg, Octoprint native support. I have read the last few days worth on the slack channel where another guy said he got one working by flashing the stock bl you have here so I re-flashed dafang to ssh in and tried that as well. I've read through all the related issues here, tried 4 different versions (0.0.24, 0.0.23, 0.0.21, and 0.0.5). I can't provide logs as I cant even get that far, and I feel like I must be missing something because it seems most people eventually get it working. Is there a way I can run it from the SD after flashing the cfw file posted [here](https://github.com/EliasKotlyar/Xiaomi-Dafang-Hacks/issues/1667#issuecomment-770461311)? I downloaded the source zip for 0.0.21 and looked through but it wasn't evident what would need to be copied for it to work (like firmware_mod from DH source) , or if it's even possible. Any guidance would be appreciated.

tachang, 2021-05-15:
Something is very odd that some of the newer V2s won't flash properly. I'm aware there are some issues going on but having a tough time tracking it down. One of the things I found was that uboot was not really flashing all the bits properly so when it tried to untar the xz file it failed on boot.

If you have DaFang working you can definitely flash it manually without uboot. However if you make a mistake you'll brick your camera.

jfsimoneau, 2021-06-12:
On my newer V2, `/proc/mtd` is different than what is written in `install.md`:
```cat /proc/mtd
dev:    size   erasesize  name
mtd0: 00040000 00008000 "boot"
mtd1: 00200000 00008000 "kernel"
mtd2: 00350000 00008000 "root"
mtd3: 000a0000 00008000 "driver"
mtd4: 004a0000 00008000 "appfs"
mtd5: 00200000 00008000 "backupk"
mtd6: 000a0000 00008000 "backupd"
mtd7: 00200000 00008000 "backupa"
mtd8: 00040000 00008000 "config"
mtd9: 00040000 00008000 "para"
mtd10: 00010000 00008000 "flag"
```

tachang, 2021-06-12:
Can you dump your uboot partition and send it over?

tachang, 2021-06-12:
It's normal for it to be different since I create a new one.

jfsimoneau, 2021-06-14:
Here's my boot partition.
[uboot.zip](https://github.com/openmiko/openmiko/files/6648960/uboot.zip)

tachang, 2021-06-15:
Thanks for providing this.

takov751, 2021-08-01:
I have purchased recently a NEOS `NS-CAM-2` in the hope I can flash it with dafang ,but i failed as @RobD83 . I follow the steps,but the first flashing process seems failed with cfw-1.2.bin, so at least my camera still not bricked, but only thing I see that it does record on the SD card with WYZE watermark.

takov751, 2021-08-13:
Since than I had time and the issue resolved When I flashed the camera with 1.1 . Since then works as expected

ramaro, 2021-09-02:
I also bought a couple of NEOS NS-CAM-2 and had the same behavior initially. However, I found the issue was that my sdcard wasn't FAT32 formatted but EXFAT instead. Formatting it with FAT32 and following the instructions worked flawlessly on both devices.

tachang, 2021-09-21:
I've pushed a new image which should solve some of this issues regarding flashing and unable to boot. The problem was not properly wiping the remnants of flash and jffs2 filesystem overrunning.

closed 2021-09-21 by tachang

RobD83, 2021-09-21:
Thank you for the update, when you first replied I wrote a lengthy reply that disappeared when I hit the green button and didn't feel like retyping it. I'm out of town right now but will try it out when I get home and hopefully it'll all be resolved. Also thanks to everyone else who commented and provided info in my stead to help with resolving the issue.

tachang, 2021-09-21:
Thanks @RobD83 . This is the release I think will fix things: https://github.com/openmiko/openmiko/releases/tag/v0.0.27