openmiko/openmiko #26: RTSP Server not working

sedony23, 2021-06-07:
Hi there,

first of all thanks for the great software. After some back and forth I got my Xiaomi Dafang flashed with the newest image (release v0.0.25). However the RTSP server doesn't seem to be working. When I try to restart the camera service I get the following:

# /etc/init.d/S60camera stop
killall: videocapture: no process killed

# /etc/init.d/S60camera start
root: Reading parameters from /etc/v4l2rtspserver.conf
Adding necessary camera modules
root: Determined image sensor model as jxf22
insmod: can't insert '/lib/modules/3.10.14/kernel/drivers/media/platform/sensors/jxf22/sensor_jxf22.ko': File exists
c
c
c
c
c
root: Starting video frame capture process
root: Waiting for initialization of v4l2 device
root: Waiting for initialization of v4l2 device
root: Waiting for initialization of v4l2 device
root: Waiting for initialization of v4l2 device
root: Waiting for initialization of v4l2 device
root: Waiting for initialization of v4l2 device
root: Unable to initialize v4l2 device
root: Starting v4l2rtspserver
VIDIOC_REQBUFS: Inappropriate ioctl for device
VIDIOC_STREAMOFF: Inappropriate ioctl for device
VIDIOC_REQBUFS: Inappropriate ioctl for device
VIDIOC_STREAMOFF: Bad file descriptor
VIDIOC_REQBUFS: Bad file descriptor
VIDIOC_STREAMOFF: Bad file descriptor
VIDIOC_REQBUFS: Bad file descriptor

I tried reflashing and with or without config files. Any suggestion?

Best

Daniel

tachang, 2021-06-29:
Odd. Can you start the videocapture process manually?

jschwalbe, 2021-09-10:
I am having the same (or similar) issue. 

```
# /etc/init.d/S60camera stop
killall: v4l2rtspserver: no process killed
# /etc/init.d/S60camera start
Adding necessary camera modules
root: Determined image sensor model as jxf23
insmod: can't insert '/driver/sensor_jxf23.ko': File exists
c
c
c
c
c
root: Starting video frame capture process
root: Starting v4l2rtspserver
VIDIOC_REQBUFS: Inappropriate ioctl for device
VIDIOC_STREAMOFF: Inappropriate ioctl for device
VIDIOC_REQBUFS: Inappropriate ioctl for device
VIDIOC_REQBUFS: Inappropriate ioctl for device
VIDIOC_STREAMOFF: Inappropriate ioctl for device
VIDIOC_REQBUFS: Inappropriate ioctl for device
# terminate called after throwing an instance of 'std::bad_alloc'
  what():  std::bad_alloc
```

Unable to connect via VLC.  Running the most current version
`Linux openmiko 3.10.14 #1 PREEMPT Mon Sep 6 21:08:31 UTC 2021 mips GNU/Linux`
on brand new Wyzecam v2.

Thoughts?

jschwalbe, 2021-09-10:
Came here to leave the same comment with the same error, but a reboot fixed it.
I dunno!
Thanks for great software.

tachang, 2021-09-21:
Closing this out since it looks like it is fixed.

closed 2021-09-21 by tachang