openmiko/openmiko #27: unable to flash fresh Xiaomi 1S

bo-skor, 2021-08-10:
Hello! Thank you for this project! 
I recently bought this camera Xiaomi 1S
my MAC beginned from E4:XX:XX:XX:XX:XX
unfortunately I could not flash it with this firmware. Also I was unable to use other projects as well.
After flashing, the LED was solid yellow, the camera was not recording. I've seen the same picture on other free projects. I would be very grateful for any help. I also find that the newer cameras are slightly different from the older versions.

schneemaier, 2021-08-14:
I have similar issue with a Xiaomi 1S. I connected to the serial port of the camera and checked what is happaning.

Based on the uboot the camera has 64 MB ram (and not 128M).

```
U-Boot SPL 2013.07 (Jul 05 2018 - 17:59:02)
pll_init:365
l2cache_clk = 375000000
pll_cfg.pdiv = 8, pll_cfg.h2div = 4, pll_cfg.h0div = 4, pll_cfg.cdiv = 1, pll_cfg.l2div = 2
nf=30 nr = 1 od0 = 1 od1 = 1
cppcr is 03c05100
CPM_CPAPCR 03b0890d
nf=42 nr = 1 od0 = 1 od1 = 1
cppcr is 02a04900
CPM_CPMPCR 07d0c90d
nf=50 nr = 1 od0 = 1 od1 = 1
cppcr is 03204900
CPM_CPVPCR 0320490d
cppcr 0x9a794410
apll_freq 712704000 
mpll_freq 1000000000 
vpll_freq = 1200000000
ddr sel mpll, cpu sel apll
ddrfreq 500000000
cclk  712704000
l2clk 356352000
h0clk 250000000
h2clk 250000000
pclk  125000000
DDRC_DLP:0000f003

U-Boot 2013.07 (Jul 05 2018 - 17:59:02)

Board: ISVP (Ingenic XBurst T20 SoC)
DRAM:  64 MiB
Top of RAM usable for U-Boot at: 84000000
Reserving 400k for U-Boot at: 83f98000
Reserving 32784k for malloc() at: 81f94000
Reserving 32 Bytes for Board Info at: 81f93fe0
Reserving 124 Bytes for Global Data at: 81f93f64
Reserving 128k for boot params() at: 81f73f64
Stack Pointer at: 81f73f48
Now running in RAM - U-Boot at: 83f98000
MMC:   msc: 0
the manufacturer c8
SF: Detected GD25Q128

*** Warning - bad CRC, using default environment

In:    serial
Out:   serial
Err:   serial
misc_init_r before change the IR_cut_gpio
gpio_request lable = IR_cut_gpio gpio = 25
misc_init_r after gpio_request the IR_cut_gpio ret is 25
misc_init_r after change the IR_cut_gpio ret is 0
misc_init_r before change the TF_CD_gpio
gpio_request lable = TF_CD_gpio gpio = 43
misc_init_r after gpio_request the TF_CD_gpio ret is 43
misc_init_r after change the TF_CD_gpio ret is 0
misc_init_r before change the SD_enable_gpio
gpio_request lable = SD_able_gpio gpio = 48
misc_init_r after gpio_request the SD_able_gpio ret is 48
misc_init_r after change the SD_able_gpio ret is 0
misc_init_r before change the wifi_reset_gpio
gpio_request lable = wifi_reset_gpio gpio = 46
misc_init_r after gpio_request the wifi_reset_gpio ret is 46
misc_init_r after change the wifi_reset_gpio ret is 0
misc_init_r before change the yellow_gpio
gpio_request lable = yellow_gpio gpio = 38
[... 114 lines cut ...]
[    0.148107] Status: 10001c03 KERNEL EXL IE 
[    0.152346] Cause : 40808008
[    0.155258] BadVA : 00000004
[    0.158174] PrId  : 00d00101 (Ingenic Xburst)
[    0.162586] Modules linked in:
[    0.165683] Process swapper (pid: 0, threadinfo=804d0000, task=804da270, tls=00000000)
[    0.173710] Stack : 00000000 00000000 00000000 803d1ec8 80590000 000002c5 804e0000 80531154
          804e0000 8052ce18 80580000 83fc4c8c 00000000 8050e4e0 804e0000 00008000
          0000012d 80582e88 00000000 00000000 804e0000 805078d8 00000000 8052ce18
          804cdbc8 00000043 ffffffff ffffffff 805072a4 00000000 80531154 00000000
          ffffffff 83f98000 83fcb894 81f73d8c 83f994c4 83f99524 00000000 00000000
          ...
[    0.209734] Call Trace:
[    0.212213] [<800cfe54>] kmem_cache_alloc+0x40/0x1c4
[    0.217252] [<8050e4e0>] pidmap_init+0x6c/0xdc
[    0.221752] [<805078d8>] start_kernel+0x344/0x3d8
[    0.226510] 
[    0.228010] 
Code: 24420001  af820014  8e110000 <8e340004> 8f820014  2442ffff  af820014  8f820008  30420004 
[    0.238121] ---[ end trace 4eaa2a86a8e2da22 ]---
[    0.242761] Kernel panic - not syncing: Attempted to kill the idle task!
[    0.249557] Rebooting in 3 seconds..
```

Flashing goes fine, but during boot it ends with a kernel panic

tachang, 2021-09-05:
64MB of ram is not supported.

schneemaier, 2021-09-06:
> 64MB of ram is not supported.

Thanks! I made a fork and adjusted some parameters to work with 64MB. However i have an issue:

```
## Booting kernel from Legacy Image at 80600000 ...
   Image Name:   Linux-3.10.14
   Image Type:   MIPS Linux Kernel Image (lzma compressed)
   Data Size:    2094520 Bytes = 2 MiB
   Load Address: 80010000
   Entry Point:  803d0730
   Verifying Checksum ... OK
   Uncompressing Kernel Image ... OK

Starting kernel ...

[    0.000000] Initializing cgroup subsys cpu
[    0.000000] Initializing cgroup subsys cpuacct
[    0.000000] Linux version 3.10.14 (root@4d23b3e78279) (gcc version 4.7.4 (Buildroot 2016.02) ) #1 PREEMPT Sun Sep 5 16:41:30 UTC 2021
[    0.000000] bootconsole [early0] enabled
[    0.000000] CPU0 RESET ERROR PC:05540428
[    0.000000] CPU0 revision is: 00d00101 (Ingenic Xburst)
[    0.000000] FPU revision is: 00b70000
[    0.000000] CCLK:712MHz L2CLK:356Mhz H0CLK:200MHz H2CLK:200Mhz PCLK:100Mhz
[    0.000000] Determined physical RAM map:
[    0.000000]  memory: 004f7000 @ 00010000 (usable)
[    0.000000]  memory: 00079000 @ 00507000 (usable after init)
[    0.000000] User-defined physical RAM map:
[    0.000000]  memory: 028b9000 @ 00000000 (usable)
[    0.000000] Initrd not found or empty - disabling initrd
[    0.000000] Zone ranges:
[    0.000000]   Normal   [mem 0x00000000-0x028b8fff]
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x00000000-0x028b8fff]
[    0.000000] Primary instruction cache 32kB, 8-way, VIPT, linesize 32 bytes.
[    0.000000] Primary data cache 32kB, 8-way, VIPT, no aliases, linesize 32 bytes
[    0.000000] pls check processor_id[0x00d00101],sc_jz not support!
[    0.000000] MIPS secondary cache 128kB, 8-way, linesize 32 bytes.
[    0.000000] Built 1 zonelists in Zone order, mobility grouping off.  Total pages: 10343
[    0.000000] Kernel command line: console=tty0 console=ttyS1,115200n8 mem=41700K@0x0 ispmem=8M@0x28B9000 rmem=15644K@0x30B9000 init=/linuxrc rootfstype=squashfs root=/dev/mtdblock2 rw mtdparts=jz_sfc:256k(boot),2048k(kernel),13504k(root),256k(config),256k(para),-(flag)
[    0.000000] PID hash table entries: 256 (order: -2, 1024 bytes)
[    0.000000] Dentry cache hash table entries: 8192 (order: 3, 32768 bytes)
[    0.000000] Inode-cache hash table entries: 4096 (order: 2, 16384 bytes)
[    0.000000] Memory: 35316k/41700k available (3877k kernel code, 6384k reserved, 1205k data, 484k init, 0k highmem)
[    0.000000] SLUB: HWalign=32, Order=0-3, MinObjects=0, CPUs=1, Nodes=1
[    0.000000] Preemptible hierarchical RCU implementation.
[    0.000000] NR_IRQS:418
[    0.000000] clockevents_config_and_register success.
[    0.000028] Calibrating delay loop... 709.42 BogoMIPS (lpj=3547136)
[    0.057699] pid_max: default: 32768 minimum: 301
[    0.062742] Mount-cache hash table entries: 512
[    0.067958] Initializing cgroup subsys debug
[    0.072222] Initializing cgroup subsys freezer
[    0.078287] devtmpfs: initialized
[    0.083234] regulator-dummy: no parameters
[    0.087594] NET: Registered protocol family 16
[    0.110606] bio: create slab <bio-0> at 0
[    0.117431] jz-dma jz-dma: JZ SoC DMA initialized
[    0.122533] usbcore: registered new interface driver usbfs
[    0.128184] usbcore: registered new interface driver hub
[    0.133640] usbcore: registered new device driver usb
[    0.138933] i2c-gpio i2c-gpio.1: using pins 57 (SDA) and 58 (SCL)
[    0.145190]  (null): set:249  hold:250 dev=100000000 h=500 l=500
[    0.151310] media: Linux media interface: v0.10
[    0.155912] Linux video capture interface: v2.00
[    0.163003] Switching to clocksource jz_clocksource
[    0.167972] cfg80211: Calling CRDA to update world regulatory domain
[    0.175632] jz-dwc2 jz-dwc2: cgu clk gate get error
[    0.180521] jz-dwc2 jz-dwc2: regulator vbus get error
[... 100 lines cut ...]
md5sum: can't open '/jffsroot/rootfs.tar.xz': No such file or directory
/bin/sh: can't access tty; job control turned off
/ # [    8.334430] jzmmc_v1.2 jzmmc_v1.2.0: card removed, state=0

```

The system mounts the /jffsroot but cannot find the file rootfs.tar.xz. When i look at demo.bin with biwalk, i can extract the jffsroot part and it has the file in it. Why the kernel is unable to mount it from flash. I tried reflashing demo.bin, flash mtd2 manually with dd, but it does not work. It mounts the partition but it seems empty:
```
/ # mount
rootfs on / type rootfs (rw)
devtmpfs on /dev type devtmpfs (rw,relatime,size=17656k,nr_inodes=4414,mode=755)
proc on /proc type proc (rw,relatime)
sysfs on /sys type sysfs (rw,relatime)
/dev/mtdblock2 on /jffsroot type jffs2 (rw,relatime)
/dev/zram0 on /newroot type ext2 (rw,relatime)
/ # ls /jffsroot/
/ # cd jffsroot/
/jffsroot # ls
/jffsroot # 
```

The only thing i changed in the source code is the kernel command line to work with 64MB:
`console=tty0 console=ttyS1,115200n8 mem=41700K@0x0 ispmem=8M@0x28B9000 rmem=15644K@0x30B9000 init=/linuxrc rootfstype=squashfs root=/dev/mtdblock2 rw mtdparts=jz_sfc:256k(boot),2048k(kernel),13504k(root),256k(config),256k(para),-(flag)`

Any help welcome. Thank you

tachang, 2021-09-06:
@schneemaier Okay I ran into this too and couldn't figure out why the file /jffsroot/rootfs.tar.xz wasn't there.

My thought was a bad flash but that doesn't seem right since I could reproduce it.
My second thought was that the flashing code only flashes some of the bytes but there is "junk" left on the other portion of the flash that the flashing code doesn't touch. When I say flashing code I mean uboot.

The reason for this is when I drop to a shell on error sometimes I see files in the /jffsroot partition that I didn't put there. They were from the factory or something. So when I remove those files it clears up the JFFS and my file shows up again.

schneemaier, 2021-09-06:
I tried unmount and remount /dev/mtdblock2 but it fails:
```
/ # mount
rootfs on / type rootfs (rw)
devtmpfs on /dev type devtmpfs (rw,relatime,size=17656k,nr_inodes=4414,mode=755)
proc on /proc type proc (rw,relatime)
sysfs on /sys type sysfs (rw,relatime)
/dev/mtdblock2 on /jffsroot type jffs2 (rw,relatime)
/dev/zram0 on /newroot type ext2 (rw,relatime)
/ # umount /jffsroot/
/ # mount
rootfs on / type rootfs (rw)
devtmpfs on /dev type devtmpfs (rw,relatime,size=17656k,nr_inodes=4414,mode=755)
proc on /proc type proc (rw,relatime)
sysfs on /sys type sysfs (rw,relatime)
/dev/zram0 on /newroot type ext2 (rw,relatime)
/ # mount /dev/mtdblock2 /mnt
[   42.118789] [EXFAT] trying to mount...
mount: mounting /dev/mtdblock2 on /mnt failed: Invalid argument
/ # mount /dev/mtdblock2 /jffsroot
[   48.455103] [EXFAT] trying to mount...
mount: mounting /dev/mtdblock2 on /jffsroot failed: Invalid argument
/ # 
```

Are there any special parameters i have to pass to make it mountable? 

I never see any files in /jffsroot, it is always empty. 

I have dumped /dev/mtdblock2 into a file on the SD card and binwalk found the partition, however it fails to extract it.

tachang, 2021-09-06:
Are you doing this via serial?

tachang, 2021-09-06:
You may want to try passing in the type of file system to the mount command.

```
# mount -t jffs2 /dev/mtdblock2 /jffstmp
# ls -al /jffstmp/
total 8523
drwxr-xr-x    3 root     root             0 Jan  1  1970 .
drwxr-xr-x   25 root     root          4096 Sep  6 20:46 ..
-rw-r--r--    1 root     root       8723340 Sep  6 19:28 rootfs.tar.xz
# 
```

schneemaier, 2021-09-06:
> Are you doing this via serial?

Yes

> You may want to try passing in the type of file system to the mount command.
This helped, however the directory is still empty...

Meanwhile i have fixed the missing `jefferson` package and extracted the bin file which i dd-ed out from the flash:
```
_mtdroot.bin-0.extracted$ tree
.
├── 0.jffs2
└── jffs2-root
    └── fs_1
        └── rootfs.tar.xz

2 directories, 2 files
```
So it is the falsh but somhow the kernel unable to mount it properly

tachang, 2021-09-06:
So you think it is the jffs mount implementation that is causing issues?

schneemaier, 2021-09-07:
Possibly, i am not at that level with kernel level programing.

In Ubuntu i use the following commands to mount the jffs2 filesystem and it works with the same image:

```
umount /dev/mtdblock0
modprobe -r mtdram
modprobe -r mtdblock
modprobe mtdram total_size=32768 erase_size=128
modprobe mtdblock
dd if="$1" of=/dev/mtdblock0
mount -t jffs2 /dev/mtdblock0 /mnt
```

schneemaier, 2021-09-15:
So i tried everything but still when the system (or I) mount with `mount -t jffs2 /dev/mtdblock2 /jffsroot` it is always empty. .If i dd /dev/mtdblock2 to the sdcard and check the file on my desktop, the filesystem and the file is there. I have no clue hwat is going wrong. Can this cause by the uboot? I still have the factory uboot from dafang on the device: U-Boot 2013.07 (Jul 05 2018 - 17:59:02)

Should i try to flash the standard Wyze v2 uboot or your custom uboot? BTW i cannot see the cimpiled uboot in the releases.

tachang, 2021-09-15:
My guess it is a bug in the jffs2 implementation I have on the kernel.
When you mount /dev/mtdblock2 can you write a file and delete it. That might work.

Feel free to join slack if you need more help.

tachang, 2021-09-15:
I wonder if I change up the jffs file system and put a few more files in there it will work around the issue.

tachang, 2021-09-15:
Do you mind sending me the full DD dump of your mtdblock2?

tachang, 2021-09-16:
One possible way to test is use jffs2 kernel implementation of 3.10.14

schneemaier, 2021-09-16:
I tried adding and deleting files, but it did not help.

This is what is see
```
Filesystem                Size      Used Available Use% Mounted on
devtmpfs                 17.2M         0     17.2M   0% /dev
/dev/mtdblock2           13.2M    392.0K     12.8M   3% /jffsroot
/dev/zram0               46.5M     20.0K     43.1M   0% /newroot
```
I noticed one more interestingthing. When i flash demo.bin, on the first boot i see the following:
```
tar: corrupted data
tar: short read
Error: Extracting root filesystem failed.
This happens on occasion from a bad flash. Try flashing again.
md5sum of /jffsroot/rootfs.tar.xz: 
848d8dea9017f4566c3c1d55679244ce  /jffsroot/rootfs.tar.xz
```
but on the second boot this is different:
```
tar: can't open '/jffsroot/rootfs.tar.xz': No such file or directory
Error: Extracting root filesystem failed.
This happens on occasion from a bad flash. Try flashing again.
md5sum of /jffsroot/rootfs.tar.xz: 
md5sum: can't open '/jffsroot/rootfs.tar.xz': No such file or directory
```

I have attache the last dump from mtdblock2 and also the demo64.bin, which is taylored for 64MB ram (different kernel command line). I have already joined slack, will ask questions there later.

[demo64.gz](https://github.com/openmiko/openmiko/files/7174108/demo64.gz)
[mtdblock2.gz](https://github.com/openmiko/openmiko/files/7174109/mtdblock2.gz)

schneemaier, 2021-09-16:
> One possible way to test is use jffs2 kernel implementation of 3.10.14

I thought the system is using kernel 3.10.14 and the jffs2 implementation is from the kernel

tachang, 2021-09-16:
So I downloaded your mtdblock2 and flashed it and tried to mount the jffs2.
I also see no files.

How big is your rootfs.tar.xz file? Maybe it is getting really close to the maximum size and because of circular writing it overwrites it? Not sure.

tachang, 2021-09-16:
Does the generated jffs2 image file match the dump you provided of mtdblock2? I want to rule out that the SPI flashing worked properly or not. I imagine the problem is not the flashing.

schneemaier, 2021-09-18:
So i bricked my device this morning. I played with uboot and probably flashed the wrong image. Will see if i can recover it wiht a SPI flasher.

To answer your qusetion: ths rootfs.tar.xz file is 8,729,776 bytes. Before is brisked the device is was playing around via serial. 

```
/sdcard/openmiko # dd if=./mtdblock2.bin of=/dev/mtdblock2  # from a working camera with openmiko
27008+0 records in
27008+0 records out
/sdcard/openmiko # md5sum mtdblock2.bin 
d1a9b98ff01bf699daadfae2bd6e154d  mtdblock2.bin
/sdcard/openmiko # dd if=/dev/mtdblock2 | md5sum
d1a9b98ff01bf699daadfae2bd6e154d  -
27008+0 records in
27008+0 records out
/sdcard/openmiko # mount -t jffs2 /dev/mtdblock2 /jffsroot
/sdcard/openmiko # md5sum rootfs.tar.xz 
2dcc60011d387a9d79773496269092b7  rootfs.tar.xz
/sdcard/openmiko # md5sum /jffsroot/rootfs.tar.xz 
2dcc60011d387a9d79773496269092b7  /jffsroot/rootfs.tar.xz
/sdcard/openmiko # cd /jffsroot/
/jffsroot # tar xvf ./rootfs.tar.xz -C /newroot/
tar: corrupted data
tar: short read
```
 then i reboted the camera
```
Starting full root buildout
Mounted core filesystems
[    6.574981] FAT-fs (mmcblk0p1): Volume was not properly unmounted. Some data may be corrupt. Please run fsck.
Mount /sdcard successful
sh: bad number
Mounted jffsroot
/jffsroot
/jffsroot/rootfs.tar.xz
[    9.965908] zram0: detected capacity change from 0 to 50331648
Filesystem label=
OS type: Linux
Block size=4096 (log=2)
Fragment size=4096 (log=2)
12288 inodes, 12288 blocks
614 blocks (5%) reserved for the super user
First data block=0
Maximum filesystem blocks=4194304
1 block groups
32768 blocks per group, 32768 fragments per group
12288 inodes per group
[    9.996845] EXT4-fs (zram0): couldn't mount as ext3 due to feature incompatibilities
[   10.008824] EXT4-fs (zram0): mounting ext2 file system using the ext4 subsystem
[   10.017902] EXT4-fs (zram0): mounted filesystem without journal. Opts: (null)
tar: corrupted data
tar: short read
Error: Extracting root filesystem failed.
This happens on occasion from a bad flash. Try flashing again.
md5sum of /jffsroot/rootfs.tar.xz: 
2dcc60011d387a9d79773496269092b7  /jffsroot/rootfs.tar.xz
/bin/sh: can't access tty; job control turned off
/ # cd /sdcard/openmiko/
/sdcard/openmiko # md5sum mtdblock2.bin 
d1a9b98ff01bf699daadfae2bd6e154d  mtdblock2.bin
/sdcard/openmiko # dd if=/dev/mtdblock2 | md5sum
f985b7a07db29194bc86ebe22d32d62b  -
27008+0 records in
27008+0 records out
```
Based on the md5sum the falsh data change during reboot. 

Notes:
1 i see that after flashing the firmware on the first boot the system complains on corrupt rootfs.tar.xz file, but on the second boot it does not even find it. 
2 the rootfs.tar.xz file works fine on my desktop. I can extract it with one warning with tar xvf command

tachang, 2021-09-18:
Dang that sucks. Can you send the rootfs.tar.xz file? Is there anything wrong with that?

schneemaier, 2021-09-18:
[rootfs.zip](https://github.com/openmiko/openmiko/files/7190697/rootfs.zip)

Here is the rootfs file. I had to zip-t otherwise github does not let me upload it

tachang, 2021-09-19:
I tried with a newer version of mkfs.jffs2. Had to compile inside container like so:

```
git clone git://git.infradead.org/mtd-utils.git
apt-get install autoconf pkgconf zlib1g-dev libtool liblzo2-dev
./autogen.sh
./configure --without-ubifs
```
The result is a new binary:

```
# /openmiko/build/buildroot-2016.02/mtd-utils/mkfs.jffs2 -V
mkfs.jffs2 (mtd-utils) 2.1.3
```

However when I use this binary vs the old one the md5sum of the resulting jffs file is the same.

```
JFFS2=$BASE_DIR/host/usr/sbin/mkfs.jffs2 
JFFS2=/openmiko/build/buildroot-2016.02/mtd-utils/mkfs.jffs2
```

This leads me to believe that jffs2 is not really an issue here. Or at least the image being generated. The flashing doesn't seem like an issue either.

My gut feeling is that because the config partition is also jffs2 the mount code is overrunning into it somehow when scanning for files. This would cause the md5sum to change. Also I am unable to reproduce it because my `/config` is working and I haven't really touched it.

```
# cat /proc/mtd 
dev:    size   erasesize  name
mtd0: 00040000 00008000 "boot"
mtd1: 00200000 00008000 "kernel"
mtd2: 00d30000 00008000 "root"
mtd3: 00040000 00008000 "config"
mtd4: 00040000 00008000 "para"
mtd5: 00010000 00008000 "flag"
```

Another reason I think this is the case is that this happens on fresh cameras so there is probably some junk jffs2 stuff sitting around at that offset which is causing interference with things.

tachang, 2021-09-19:
For stock wyzecams the way the uboot is configured it flashes the following:

```
# Maximum sizes
#                                                                                                                                                                      
# "kernel" - 0x200000 or 2,097,152 bytes
# "rootfs" - 0x350000 or 3,473,408 bytes
# "driver" - 0xa0000 or 655,360 bytes
# "appfs" - 0x4a0000 or 4,849,664 bytes
```

I am "cheating" in that I keep the kernel the exact same size but I treat rootfs/driver/appfs as all one big binary. uboot doesn't really know that so it just happily keeps writing the new firmware. However there is no guarantee that after appfs there is a clean file system. But shouldn't mtd stop us from reading past that?

tachang, 2021-09-21:
I finally figured this out. This is the commit that fixes the issue:

https://github.com/openmiko/openmiko/commit/45bc5ad9f46af375bf14e5edae551b71a5b8a516

closed 2021-09-21 by tachang

bikemike, 2021-11-26:
> I finally figured this out. This is the commit that fixes the issue:
> 
> [45bc5ad](https://github.com/openmiko/openmiko/commit/45bc5ad9f46af375bf14e5edae551b71a5b8a516)

Does this mean that openmiko can now support 64MB flash?

@schneemaier Did you ever unbrick your camera and what changes did you have to make to get it working with 64Mb flash?

schneemaier, 2021-11-27:
@bikemike No 64MB needs a different firmware. The currernt fimrware uses a 48MB ram disk (zram) for the rootFS, this does not work for the 64MB camera as the whole operating memory (after sensor reservations) is around 35MB.

Also while the fix resolved the issue with the jffs2 partition mount on the 128MB camera i still get a corrupted tar file on my 64MB version when i am testing. I did not dig into the issue deeper as i have other things to do now. 

Recovering a 64MB xiaofang 1S camera i searhed the net for the fimrware (as i recall i found it on a chinees website) and flashed it as demo.bin. UNfortunatelly i don't have that file anymore, but you can find the link and process here: https://c.mi.com/thread-2106610-1-0.html

Note: that i was unable to use the camera with the original firmware as somehow it refused to configure (maybe due to some regon lock i read about in the forums)

iz3man, 2024-01-16:
How do I find out what memory size my Xiaomi Xiaofang 1S is? Could it be that it's only 32MB?
Looks like 32MB to me, so I guess installing this firmware is not possible, right?

[image]

schneemaier, 2024-01-17:
> How do I find out what memory size my Xiaomi Xiaofang 1S is? Could it be that it's only 32MB? Looks like 32MB to me, so I guess installing this firmware is not possible, right? [...]

You are better of to check `cat /proc/cmdline`

Example for the 128MB version:
```
yacam:~# cat /proc/cmdline
console=tty0 console=ttyS1,115200n8 mem=96M@0x0 ispmem=8M@0x6000000 rmem=24M@0x6800000 init=/linuxrc rootfstype=squashfs root=/dev/mtdblock2 rw mtdparts=jz_sfc:256k(boot),2048k(kernel),8768k(root),4736k(extra),256k(config),256k(para),-(flag)
```

Total memory = mem + ispmem + rmem = 96M + 8M +24M = 128M

iz3man, 2024-01-17:
Thanks. So quite disappointing, and for sure the reason I couldn't install the FW: mem:32Mb +ispmem:20Mb + rmem:24Mb = 64Mb :(

schneemaier, 2024-01-23:
> Thanks. So quite disappointing, and for sure the reason I couldn't install the FW: mem:32Mb +ispmem:20Mb + rmem:24Mb = 64Mb :( I noted that you mentioned starting a fork for 64Mb versions. But I'm unsure if you've been able to make it work?! Is that YACAM? But I read it's working/tested for 128Mb only?! Should I give it a try? EDIT: Just read the release notes. The latest 64Mb has comment : Video not working, and there is no new one. So no sense trying that ;)

I have a 64MB version where the OS boots and the camera is reachable. However I am unable to get the camera working as the `[ingenic_videocap]` binary exists with illegal instruction. Unfortunatelly I had no time to debug it and figure out why it is failing.