openmiko/openmiko #28: How do you set timezone?

sflitman, 2021-08-22:
I have a running WyzeCam2 with release 0.0.23, it is set at UTC for the on-screen display on the video stream.  How do I set an alternative timezone?
Thanks,
SSF

tachang, 2021-09-05:
Saw this posted on the Slack channel:

For those looking to understand how to set your timezone, there are instructions at this page: https://pubs.opengroup.org/onlinepubs/007904975/basedefs/xbd_chap08.html

The are further examples here: http://leaf.sourceforge.net/doc/buci-tz.html

As an example for my situation (Eastern Timezone), I've done the following:
echo EST5EDT > /etc/TZ # assigns the value now...
echo EST5EDT > /sdcard/config/overlay/etc/TZ # assign the value for reboots...

tachang, 2021-09-05:
For pacific time:

PDT8PST

sflitman, 2021-09-06:
Fantastic!   I was in Phoenix which lacks DST, so the correct string is MST7.
Best,
SSF

closed 2021-09-06 by sflitman

peliopoulos, 2022-01-05:
I did the PDT8PST entry, but I'm 12 hours behind.  Its showing 03:41 (no AM or PM designaton) when it should be 15:41.  I'm on the latest firmware version.