openmiko/openmiko #29: Unable to flash iSmartAlarm Spot+ (iSC5P)

bediston, 2021-08-31:
Hello,
I tried a lot of times to flash openmiko on iSmartAlarm Spot+ (iSC5P).
I tried different ways and steps, different sd card (with different partitions), pushing on "setup" while plugin power for 1/2/5/10/15/20/30... seconds.

I still have no blue led, only an orange one blinking fast.
And if I stay more than 30s on setup, I have a voice telling me "Ready to connect".

Same behavior with version of "Xiomi-Dafang-Hacks".

Anybody managed to flash an iSmartAlarm Spot+ or can I just throw them to the bin?

Thank you for your help.

tachang, 2021-09-06:
How do you typically flash it? When it says Ready to Connect doesn't that mean it is default firmware?

rubin110, 2022-11-14:
If I had to guess, I would suspect whatever is happening here has nothing to do with this open source firmware, but worth documenting for future generations.

I'm having a similar issue with a `WYZEC2BK`, `2C:AA:8E:92:2F:19`. Holding down the setup button does not trigger the blue LED. I've tried a few different SD cards, 8gb, 32gb. Formatted either through the camera or repartitioning and formatting exfat through Linux, with the file named `demo.bin`. Can't seem to get it to switch over to the blue LED.

I can however reset and pair the camera with the app just fine. Blue LED shines when it's gotten on wifi. Wyze firmware `4.9.9.1472`.

If I still had the RTSP firmware I would try flashing that.

Also hey Jeff, long time no see. Feel free to email me if you want to catch up.

rubin110, 2022-11-14:
Formatting with FAT32 did the trick.