openmiko/openmiko #30: Unable to Stream Video with openmiko.conf ENABLE_AUTONIGHT=0

rjgrandy, 2021-09-01:
I am able to start the camera with the default settings. However due to the nightvision bug where it will get stuck in the mode, I wanted to disable night vision. I added the openmiko.conf file into the /config/overlay/etc directory with the modification to the enable_autonight line. After starting the camera, the stream never comes available. I can revert back to the default setup and everything will work again.

I see the following error over and over again in the log file:

```
Aug 31 07:27:04 openmiko local1.err videocapture[325]: 1921467632: Buffer underrun when writing to ALSA loopback device
Aug 31 07:27:04 openmiko local1.err videocapture[325]: 1921467632: ERROR. Can't write to PCM device. Broken pipe
```

However I can ssh into the camera.

tachang, 2021-09-06:
Sometimes if you reboot a few times it will work. I pushed a firmware trying to get audio to work but sometimes due to race conditions it doesn't come up. So I am pushing a new one soon that allows you to disable audio until I get it working.

rjgrandy, 2021-09-10:
Great I'll try the update to see if it works. I don't need audio so if disabling it fixes the autonight issue then it will be gravy!