openmiko/openmiko #31: How can I make the on-screen time display 24 hour?

sflitman, 2021-09-06:
I see here in [sample-OSD.c](https://github.com/openmiko/openmiko/blob/d4bc71a22556bbbe2f68528b9a0f82dfd4d2fd6e/custompackages/package/ingenic_samples/libimp-samples/sample-OSD.c#L92) that %I is used instead of %H in the call to strftime, maybe that could be made a configuration option?
Best,
SSF

tachang, 2021-09-06:
Definitely. I just hard code it here for now:

https://github.com/openmiko/ingenic_videocap/blob/master/src/capture.c#L631

I had a lot trouble understanding how the code actually works with drawing it on the framebuffer so still have to go through and understand a bit of it.

But can definitely make it configurable.

schneemaier, 2021-09-14:
As the time is in 12h format now it would beneficial to change line 631 in `capture.c` to `strftime(DateStr, 40, "%Y-%m-%d %I:%M:%S %p", currDate);` to include AM / PM or to `strftime(DateStr, 40, "%Y-%m-%d %H:%M:%S", currDate);` to make it 24h.

tachang, 2021-10-05:
@sflitman This feature now in the latest release. You can thank @schneemaier for his contribution!

-Jeff

closed 2021-10-05 by tachang