openmiko/openmiko #32: No video in latest

schneemaier, 2021-09-14:
Hi,

installed the latest openmiko on a wyze v2 camera. The camera boots, but i had to make some tweaks to the wifi config file. I had to comment out  pairwise and group lines to connect to my wifi:
```
# This file is used to configure wifi
#
# For more configuration option please see:
# https://w1.fi/cgit/hostap/plain/wpa_supplicant/wpa_supplicant.conf

ctrl_interface=/var/run/wpa_supplicant
ctrl_interface_group=0
ap_scan=1

network={
        ssid="******"
        # Uncomment to connect to Hidden SSIDs
        #scan_ssid=1 
        key_mgmt=WPA-PSK
        # pairwise=CCMP TKIP
        # group=CCMP TKIP WEP104 WEP40
	# psk stands for pre-shared key (the wifi password)
        psk="******"
        priority=2
}
```

But even after connecting the video stream does not work and mjpg_stream does not start.

here is the videocapture_settings.json:
```
"general_settings": {
	"flip_vertical": 0,		// 1 flips image along vertical axis, 0 disables
	"flip_horizontal": 0,		// 1 flips image along horizontal axis, 0 disables
	"show_timestamp": 1, 		// 1 enables timestamp, 0 disables
	"enable_audio": 0
},
```

VLC can connect to rtsp://IP:8554/video3_unicast but no video or audio, cannot connect to rtsp://IP:8554/video5_unicast

mjpg stream seems to start, but no mjpg process and cannot connect to the web stream:
```
# ps | grep mjpg
 8677 root       0:00 grep mjpg
# /etc/init.d/S70mjpg_streamer start
Starting mjpg_streamer (15 fps): OK
# /etc/init.d/S70mjpg_streamer stop
Stopping mjpg_streamer: killall: mjpg_streamer: no process killed
FAIL
# mjpg_streamer -i "input_v4l2loopback.so -d /dev/video4 --fps 15" -o "output_http.so"
MJPG Streamer Version.: 2.0
Error setting v4l2_format using VIDIOC_S_FMT
: Device or resource busy
input_init() return value signals to exit
# 
```

Any idea what am i doing wrong?

Thanks

schneemaier, 2021-09-14:
Okay, i figured out that my videocapture_settings.json was incomplete. The installation quide was somehow misleading. Now i have mjpg_streamer up and running, but no rstp stream. Here is the log:

```
Sep 14 01:35:35 openmiko local1.notice videocapture[335]: 1996686128: Added the jxf23 sensor.
Sep 14 01:35:35 openmiko kern.debug kernel: [   34.844104] do not support ctrl->cmd ====1
Sep 14 01:35:35 openmiko kern.debug kernel: [   34.844977] jxf23 stream on
Sep 14 01:35:35 openmiko kern.warn kernel: [   34.870142] ###### image_tuning_v4l2_open 4329 #######
Sep 14 01:35:35 openmiko local1.notice videocapture[335]: 1996686128: Sensor succesfully initialized.
Sep 14 01:35:35 openmiko local1.notice videocapture[335]: 1996686128: Loading general settings
Sep 14 01:35:35 openmiko local1.notice videocapture[335]: 1996686128: general_settings:   flip_vertical: 0 flip_horizontal: 0 show_timestamp: 1 enable_audio: 0
Sep 14 01:35:35 openmiko local1.notice videocapture[335]: 1996686128: Loading frame sources
Sep 14 01:35:35 openmiko local1.notice videocapture[335]: 1996686128: Found 2 frame source(s).
Sep 14 01:35:35 openmiko local1.notice videocapture[335]: 1996686128: FrameSource:  id: 0 group: 0 pixel_format: 10 pixel_format_name: PIX_FMT_NV12 frame_rate_numerator: 25 frame_rate_denominator: 1 pic_width: 1920 pic_height: 1080
Sep 14 01:35:35 openmiko local1.notice videocapture[335]: 1996686128: Created frame source channel 0
Sep 14 01:35:35 openmiko local1.notice videocapture[335]: 1996686128: Frame source setup complete
Sep 14 01:35:35 openmiko local1.notice videocapture[335]: 1996686128: FrameSource:  id: 1 group: 1 pixel_format: 10 pixel_format_name: PIX_FMT_NV12 frame_rate_numerator: 25 frame_rate_denominator: 1 pic_width: 640 pic_height: 360
Sep 14 01:35:35 openmiko local1.notice videocapture[335]: 1996686128: Created frame source channel 1
Sep 14 01:35:35 openmiko local1.notice videocapture[335]: 1996686128: Frame source setup complete
Sep 14 01:35:35 openmiko local1.notice videocapture[335]: 1996686128: Loading encoders
Sep 14 01:35:35 openmiko local1.notice videocapture[335]: 1996686128: Found 3 encoders.
Sep 14 01:35:35 openmiko local1.notice videocapture[335]: 1996686128: Setting up encoder attributes
Sep 14 01:35:35 openmiko local1.notice videocapture[335]: 1996686128: Done setting up encoder attributes
Sep 14 01:35:35 openmiko local1.notice videocapture[335]: 1996686128: EncoderSetting:  channel: 0 group: 0 v4l2_device_path: /dev/video3 payload_type: PT_H264 profile: 0 mode: ENC_RC_MODE_H264VBR frame_rate_numerator: 25 frame_rate_denominator: 1
Sep 14 01:35:35 openmiko local1.notice videocapture[335]: 1996686128: Encoder channel attributes for channel 0
Sep 14 01:35:35 openmiko local1.notice videocapture[335]: 1996686128: IMPEncoderCHNAttr:  payload_type: PT_H264 bufSize: 0 profile: 0 picWidth: 1920 picHeight: 1080 attrH264Vbr.outFrmRate.frmRateNum: 25 attrH264Vbr.outFrmRate.frmRateDen: 1 attrH264Vbr.maxGop: 10 attrH264Vb
Sep 14 01:35:35 openmiko local1.notice videocapture[335]: 1996686128: Created encoding group 0
Sep 14 01:35:35 openmiko local1.notice videocapture[335]: 1996686128: Created encoder channel 0
Sep 14 01:35:35 openmiko local1.notice videocapture[335]: 1996686128: IMP_Encoder_RegisterChn(Group [0], Channel [0])
Sep 14 01:35:35 openmiko local1.notice videocapture[335]: 1996686128: Setting up encoder attributes
Sep 14 01:35:35 openmiko local1.notice videocapture[335]: 1996686128: Done setting up encoder attributes
Sep 14 01:35:35 openmiko local1.notice videocapture[335]: 1996686128: EncoderSetting:  channel: 1 group: 0 v4l2_device_path: /dev/video4 payload_type: PT_JPEG profile: 0 mode: ENC_RC_MODE_H264VBR frame_rate_numerator: 25 frame_rate_denominator: 1
Sep 14 01:35:35 openmiko local1.notice videocapture[335]: 1996686128: Encoder channel attributes for channel 1
Sep 14 01:35:35 openmiko local1.notice videocapture[335]: 1996686128: IMPEncoderCHNAttr:  payload_type: PT_JPEG bufSize: 0 profile: 0 picWidth: 1920 picHeight: 1080 attrH264Vbr.outFrmRate.frmRateNum: 25 attrH264Vbr.outFrmRate.frmRateDen: 1 attrH264Vbr.maxGop: 10 attrH264Vb
Sep 14 01:35:35 openmiko local1.notice videocapture[335]: 1996686128: Created encoding group 0
Sep 14 01:35:35 openmiko local1.notice videocapture[335]: 1996686128: Created encoder channel 1
Sep 14 01:35:35 openmiko local1.notice videocapture[335]: 1996686128: IMP_Encoder_RegisterChn(Group [0], Channel [1])
Sep 14 01:35:35 openmiko local1.notice videocapture[335]: 1996686128: Setting up encoder attributes
Sep 14 01:35:35 openmiko local1.notice videocapture[335]: 1996686128: Done setting up encoder attributes
Sep 14 01:35:35 openmiko local1.notice videocapture[335]: 1996686128: EncoderSetting:  channel: 2 group: 1 v4l2_device_path: /dev/video5 payload_type: PT_H264 profile: 2 mode: ENC_RC_MODE_H264VBR frame_rate_numerator: 25 frame_rate_denominator: 1
Sep 14 01:35:35 openmiko local1.notice videocapture[335]: 1996686128: Encoder channel attributes for channel 2
Sep 14 01:35:35 openmiko local1.notice videocapture[335]: 1996686128: IMPEncoderCHNAttr:  payload_type: PT_H264 bufSize: 0 profile: 2 picWidth: 640 picHeight: 360 attrH264Vbr.outFrmRate.frmRateNum: 25 attrH264Vbr.outFrmRate.frmRateDen: 1 attrH264Vbr.maxGop: 10 attrH264Vbr.
Sep 14 01:35:35 openmiko local1.notice videocapture[335]: 1996686128: Created encoding group 1
Sep 14 01:35:35 openmiko local1.notice videocapture[335]: 1996686128: Created encoder channel 2
[... cut ...]
Sep 14 01:36:45 openmiko local1.notice videocapture[335]: 1901372656: Current FPS: 25.00 / Channel 2
Sep 14 01:36:49 openmiko local1.notice videocapture[335]: 1905566960: Current FPS: 25.00 / Channel 0
Sep 14 01:36:53 openmiko local1.notice videocapture[335]: 1901372656: Current FPS: 25.00 / Channel 2
Sep 14 01:36:53 openmiko local1.notice videocapture[335]: 1903469808: Current FPS: 25.00 / Channel 1
Sep 14 01:36:57 openmiko local1.notice videocapture[335]: 1905566960: Current FPS: 25.00 / Channel 0
```

schneemaier, 2021-09-14:
Few reboots later rstp stream is working. Wierd 

Closing

closed 2021-09-14 by schneemaier

yebo29, 2021-11-05:
@schneemaier can you please share your videocapture_settings.json file? I'm facing a similar issue.

schneemaier, 2021-11-07:
@yebo29 Sorry i don't have that config file, as i reflashed the camera several times synce.

The important part is that you have to have every possible  option set in the file otherwise it will fail to load. So start with the file provided for the ingenic videocapture: https://github.com/openmiko/ingenic_videocap/blob/master/settings.json. Don't forget to rename it to videocapture_settings.json and place it into the correct folder on the SD Card

MetricsGaming, 2022-02-01:
Does anyone have a complete videocapture_settings.json they could post? I've copy pasted from the OP's link above and renamed. Getting errors in videocapture.log
 ```
23:52:49 ERROR /openmiko/build/buildroot-2016.02/output/build/ingenic_videocap-f56caf60df55b16f9825e214183b567cd34aaaee/src/main.c:494:2003071792: Unable to parse JSON data
23:52:49 ERROR /openmiko/build/buildroot-2016.02/output/build/ingenic_videocap-f56caf60df55b16f9825e214183b567cd34aaaee/src/main.c:500:2003071792: Error before: ]
```
and in messages log
```
Jan 31 23:53:04 openmiko user.notice root: Waiting for initialization of v4l2 device
Jan 31 23:53:05 openmiko user.notice root: Unable to initialize v4l2 device
```

gleep52, 2022-02-05:
@MetricsGaming same issues here - you figure this out?

gleep52, 2022-02-06:
Found my answer here: https://github.com/openmiko/openmiko/pull/5

It appears to be a totally different json file in the minimal directory which doesn't have the parsing error and properly starts up.