openmiko/openmiko #50: Wyzecam V3?

jplwoodward, 2022-02-14:
Any plans to support Wyzecam V3?

blakeasullivan, 2022-04-05:
Hopefully I'm not giving too much away, but Wyze still sells the V2 in Black on their website (much cheaper than the sellers on eBay), FWIW https://www.wyze.com/products/wyze-cam-black

minghuadev, 2022-05-29:
+1 for v3. Anybody has any plan? 

A few links about v3: 

https://github.com/openmiko/openmiko/issues/50 - possible UART pins
https://github.com/HclX/WyzeHacks/issues/102 - where to get toolchain
https://github.com/HclX/WyzeHacks/issues/146 - initial setup without internet possible?
https://github.com/gtxaspec/wz_mini_hacks/issues/38 - internet required?

https://www.mouser.com/ProductDetail/Lumissil/T31ZX?qs=ljCeji4nMDlCcKictkO8%2Fw%3D%3D - Ingenic T31ZX on mouser with a brief datasheet

beankylla, 2022-07-31:
@tachang  would you consider expanding firmware to v3 if we provide you a V3 wyzecam? would be really happy to contribute