openmiko/openmiko #51: Night Mode

jamesleemarsh, 2022-02-27:
when Wyze v2 using Openmiko automatically turns to night mode image turns to grey scale. When manual activation of night mode via curl the image is purple. 

Also Auto night mode is still not 100% seems to work when first boot but after hours of use the auto night mode stops working. and if activated overnight doesn't like to auto deactivate night mode during high ambient light.

tachang, 2022-03-06:
Thanks for this report. I am updating the autonight script values with ones from dafang in this next version.

How are you activating night mode two different ways? Both ways seem to go through the same function right?

jamesleemarsh, 2022-03-06:
Im mostly using the curl commands. But when the camera is first turned on the automatic night mode works for the first 10-60min. After that it has issues turning on or turning off auto night mods.

The curls commands work anytime but suffers from the purple color issue.

WimVDK, 2022-03-24:
I'm currently running v0.0.38 and i personally like the greyscale mode. It is my preferred mode, that seems to have vanished after updating from 0.0.31 to 0.0.38. On top of that, the nightmode script seems "glitchy" it often toggles between day and night (when watching my 3D printer) multiple times in a matter of minutes, and seems to get "stuck" in day or night mode after a while too. Only fix when that happens is to restart it by pulling power

tachang, 2022-03-24:
@WimVDK Have you tried v0.0.40?

WimVDK, 2022-03-24:
No I haven't, i don't check daily on new versions. Can't do it right now. Running a print. It'll be for tomorrow.

Slight sidenote: haven't been able to get rtsp working either. I have a USB A to A cable but not sure how to move from there. I'd be happy to post an 'issue' if you want. (Not sure where to look if it's a WyzeCam v2 issue, or something missing on the raspberry or octoprint side)

schneemaier, 2022-03-28:
Noghtmode stops working for me after a few hours too. I tried to debug and here is my experience:
1. stop, starting or restarting autonight service does not fix the issue
2. The change between BW and COLOR camera picture still works
3. The IR leds are staying on and the IR filter also 

Tried to manualy change the IR LED and IR filter status, but it does not work. Command returns an error:

```
# curl -d value=1 http://127.0.0.1:8081/api/ir_cut
curl: (52) Empty reply from server
# curl -d value=0 http://127.0.0.1:8081/api/ir_led
curl: (52) Empty reply from server
```

Hope this helps

WimVDK, 2022-04-10:
on a sidenote, is there an easier way to upgrade the firmware than the whole "tiny button holding" method? My clock is now also behind 1hr due to switching to summertime in the last weekend of march? What would be the proper overlay settings for Belgium DST & non-DST?

bill-mcgonigle, 2022-04-19:
@WimVDK have a look at #56 for upgrading.

WimVDK, 2022-04-19:
https://github.com/openmiko/openmiko/issues/63

symphonymarie, 2022-09-23:
> Noghtmode stops working for me after a few hours too. I tried to debug and here is my experience:
>
>     1. stop, starting or restarting autonight service does not fix the issue
>
>     2. The change [...]

Did you ever solve this? I have this issue as well, same error.

schneemaier, 2022-12-24:
Has some time and created a workaround. It seems the problem is around the app.py webservice which dies after some time. Instead of that i now directly adjust the gpio pins from the nigthmode.sh script which works so far