openmiko/openmiko #54: How to set the timezone

jbeck22, 2022-03-24:
How do you set the timezone / time?  Currently my camera is showing 06:05:02, but locally it is 2:05pm.

Sparticuz, 2022-03-31:
You'll want to add an overlay. https://github.com/openmiko/openmiko/wiki/Setting-the-Time-Zone

BNWEIN, 2022-04-21:
Cant for the life of me work out how to do this either! 

I tried creating a file called TZ in /etc and i entered "Europe/London" which i believe is the TZ Database name for my location, and i restarted the time service, but no change?

Sparticuz, 2022-04-21:
You need to use the other format. I believe for London it will be `GMT0BST`, and you will want to put it on `/sdcard/config/overlay/etc/TZ`, then you can reboot the camera (probably easier than just the time service)