openmiko/openmiko #55: Low quality stream & Night mode fixes

eebrains, 2022-03-25:
For Wyze V2:

I did lots of experiments and came up with these overlay files (put this on the sd card and reboot)
[config.zip](https://github.com/openmiko/openmiko/files/8354072/config.zip)

Fixes the night  mode issues I had. (YMMV)
High quality stable stream over wifi.  (reduced to 720p, 10FPS  - but see for yourself - its good)

Put them here to help others.

tachang, 2022-03-26:
Cool thank you for the contribution. Yeah this reduces the resolution so I assume the performance would be better.

jschwalbe, 2022-04-01:
@eebrains thanks for this! Question I'm trying to figure out, I set `"show_timestamp": 1` and now the time OSD is blinking pretty rapidly/inconsistently. Any ideas? Thanks!

https://user-images.githubusercontent.com/8005921/161179674-8bc108b1-cb97-4bf8-b6e0-a6e700293135.mov

symphonymarie, 2022-04-24:
Used your config files and loving it, thanks! And thanks to the openmiko team!

Shareisfun, 2022-05-01:
My stream worked, but was not happy with the night shifting problem. I put your files on my sd card and now my stream dosent worke. I can se that it has the same ip but no stream http://192.168.1.144:8080/?action=stream

symphonymarie, 2022-06-07:
Question about the night mode - 
`/usr/bin/autonight -vv -S -c /usr/bin/nightmode.sh -j 3 -w 3 -1 1220000 -2 930000,18,1 -3 2500,17,1`
What do these numbers mean or refer to? Is it ISO?
I wanted to adjust the night mode trigger as we just got an outdoor security light, and the possums are constantly setting it off making fine-tuning difficult lol.

martinbutt, 2022-07-24:
For some reason these settings now trash the camera, and RTSP doesn't come up. Looks like they conflict with some commits in May. I didn't try to diagnose the issue much, but seems the file system gets stuck in read-only mode. I've had this happen on two cameras with completely fresh installs. The only way I've found to make the camera usable again is to format the SD card, hard reset the camera (hold button until blue light flashes 3 times), and reflash with demo.bin. 

After doing some digging, it looks like the commits from May also bring in some of these tweaks for night mode, so this may not even be necessary anymore.