openmiko/openmiko #56: Upgrade Firmare

ehm9570, 2022-03-29:
Going back through the releases I found the upgrade procedure in v29.  Might be nice to show on installation page.
For anyone looking command is (over ssh):

upgrade_openmiko v0.0.xx

xx is version number.

I used 'upgrade_openmiko.sh v0.0.40'  to get v40.

jschwalbe, 2022-03-30:
Assuming you do this first:

```
curl https://raw.githubusercontent.com/openmiko/openmiko/master/overlay_minimal/usr/bin/upgrade_openmiko.sh > upgrade_openmiko.sh
chmod +x upgrade_openmiko.sh
```

bill-mcgonigle, 2022-04-19:
Folks, I typed up your code into a pull request 'for the next guy' .  Have a look if you can.  Thanks for helping me upgrade my cameras.

WimVDK, 2023-01-28:
Trying to follow this procedure to upgrade to v0.41 alpha 2. Stuck on line 2
xauth not found when trying to chmod +x
> root@192.168.1.111: curl https://raw.githubusercontent.com/openmiko/openmiko/master/overlay_minimal/usr/bin/upgrade.sh > upgrade_openmiko.sh
> sh: /usr/bin/xauth: not found
>   % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
>                                  Dload  Upload   Total   Spent    Left  Speed
> 100    14  100    14    0     0     42      0 --:--:-- --:--:-- --:--:--    47
> root@192.168.1.111:# chmod +x upgrade_openmiko.sh
> sh: /usr/bin/xauth: not found
> root@192.168.1.111:#

martinbutt, 2023-03-24:
@WimVDK this is because your SSH shell has X11 Forwarding enabled. This could be in your `~/.ssh/config`, or another setting in your SSH client.

WimVDK, 2023-08-27:
Now that energy prices have come down in my area to almost pre-covid levels i wanna get abck into printing. That means bringing my cam fw up-to-date. My client is currently SmarTTY Portable. I can't find the setting to disable X11 forwarding. Any idea's where to look or if the installed version would have that option (or an alternate easy to use ssh client where i can disable it)?

WimVDK, 2024-12-01:
@martinbutt @jschwalbe since i'm rebuilding my cr10 max into a klipper machine i thought i'd give this a try again. Can't imagine i never thought of this before but i just used windows command prompt to ssh into the webcam. Pasted the command

curl https://raw.githubusercontent.com/openmiko/openmiko/master/overlay_minimal/usr/bin/upgrade_openmiko.sh > upgrade_
openmiko.sh

which gave me an output of

  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100  2836  100  2836    0     0   6725      0 --:--:-- --:--:-- --:--:--  7562

Then followed by chmod +x upgrade_openmiko.sh

which gave me nothing at all. How am i supposed to know now if the upgrade worked or not? I tried to run "upgrade_openmiko.sh" but that just gives me "-sh: upgrade��_openmiko.sh: not found"

martinbutt, 2024-12-07:
I've never used that script myself, so I can vouch for if it works. When you try to run it, are you using `./upgrade_openmiko.sh`, with the dot slash at the beginning? The dot slash is required on *nix operating systems.

WimVDK, 2024-12-24:
@martinbutt sorry for the delay. Busy klipperizing my cr10 max

# ./upgrade_openmiko.sh
-sh: ./upgrade_openmiko.sh: Permission denied
# sudo ./upgrade_openmiko.sh
sudo: ./upgrade_openmiko.sh: command not found

WimVDK, 2024-12-24:
ok so apparently i needed to do 

`chmod +x upgrade_openmiko.sh`

and then did 

`sudo ./openmiko_upgrade` which gave me 

`./upgrade_openmiko.sh <GIT_TAG>`. 
So first i tried` ./upgrade_openmiko.sh 0.0.40`" which led to 

```
Are you sure you want to flash new firmware? y

Stopping services...
Services stopped.

Downloading updated images from GitHub...
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
curl: (22) The requested URL returned error: 404 Not Found
Failed to download OpenMiko 0.0.40 rootfs.jffs2 from GitHub.

```

So figured ok, the release tags say v0.0.40. So lets try that, which led to 

```
# sudo ./upgrade_openmiko.sh v0.0.40
Are you sure you want to flash new firmware? y

Stopping services...
killall: mjpg_streamer: no process killed
Process was not running.
killall: autonight: no process killed
Process was not running.
killall: videocapture: no process killed
Process was not running.
killall: v4l2rtspserver: no process killed
Process was not running.
Services stopped.

Downloading updated images from GitHub...
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
100 8678k  100 8678k    0     0  2509k      0  0:00:03  0:00:03 --:--:-- 3796k
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
100 2046k  100 2046k    0     0  1109k      0  0:00:01  0:00:01 --:--:-- 1543k
Images downloaded successfully.

Flashing root file system (rootfs)
Flashing contents of rootfs.jffs2 to /dev/mtd2...
Erasing 32 Kibyte @ d30000 - 100% complete.
17357+1 records in
17357+1 records out
Wrote rootfs.jffs2 to /dev/mtd2.
Comparing contents of /dev/mtd2 to rootfs.jffs2... this may take a while
Verified the contents of /dev/mtd2 match the source file /tmp/rootfs.jffs2.

Flashing kernel image (uImage.lzma)
Flashing contents of uImage.lzma to /dev/mtd1...
Erasing 32 Kibyte @ 200000 - 100% complete.
4093+1 records in
4093+1 records out
Wrote uImage.lzma to /dev/mtd1.
Comparing contents of /dev/mtd1 to uImage.lzma... this may take a while
Verified the contents of /dev/mtd1 match the source file /tmp/uImage.lzma.

OpenMiko image flashed successfully. Reboot is required.
```

Finally success! Took a while to figure this out, the documentation could use an update

Full upgrade instructions

1. ssh root@ip
2. enter password: root
3. curl https://raw.githubusercontent.com/openmiko/openmiko/master/overlay_minimal/usr/bin/upgrade_openmiko.sh > upgrade_openmiko.sh
4. chmod +x upgrade_openmiko.sh
5.  ./upgrade_openmiko.sh <version number> aka "./upgrade_openmiko.sh v0.0.40"

Is there any way to know via ssh or otherwise which firmware version is present on the cam.

Greyscale autotoggle night vision remains bugged