openmiko/openmiko #60: dead v2 Cam

EQNish, 2022-04-06:
I need a little help, I uploaded the OpenMiko firmware and all was good until I attempted to edit the videocapture_settings.json.
after my edit; I lost video feed and could not get it back, I attempted to reload the firmware (using the same process as the original load, and now I get no response from the Cam, the light on back will only flash if no SD card is installed, and I get no IP response... I believe the unit is bricked!

Also of note, I attempted to reload the "default" firmware as provided by this group/page to no avail!  please provide some kind of feed back, I attempted to get help in the slack, but there seems to be little or no action there!

marcobe, 2022-04-06:
Maybe try another SD card?

EQNish, 2022-04-08:
no SD card is fine, works in other Cams