openmiko/openmiko #62: Cam stops working after some time

Nordikk, 2022-04-14:
I'm using OpenMiko on my WyzeCam v2.

After some time (not always the same time, it happens seamingly randomly) it just "crashes", but not completely. 
Just the streams and the integrated "web server" stops working, so via browser or VNC I can't connect anymore.

SSH stills works fine though, so I have to connect to ssh every few hours / days and reboot it, then it'll work fine for some time.

jakest, 2022-04-19:
This is happening to me as well. It seems completely random.

bill-mcgonigle, 2022-05-07:
Poking around on ssh, nothing was listening on 8554.  I rebooted and it was listening again, but no picture showed up on my monitor.  To troubleshoot I ran:

```
mplayer -rtsp-stream-over-tcp rtsp://camera:8554/video3_unicast
```

and that worked, and at the same time the picture showed up on my monitor.  As if whatever mplayer did with the server convinced it to serve other clients.  Dunno why that might be.

symphonymarie, 2022-05-09:
This issue just started cropping up for me as well. I thought maybe it was a power issue, but doesn't seem to be the case. Same issue on two cams. Connected to the network just fine, ssh works.

bill-mcgonigle, 2022-05-14:
My previous comment was probably just a coincidence, so ignore that.  There's no process listening on 8554.  http://IP:8080/?action=stream does work so the camera is up. 

Restarting S60camera doesn't do it, but manually starting:
```
/usr/bin/v4l2rtspserver -A 48000 -C 1 -a S16_LE -l 1 /dev/video3,hw:0,1 /dev/video5
```
does.  I'm leaving it running from an attached ssh session with `-v` to see if I get any error messages.

symphonymarie, 2022-05-17:
As an update to this issue, one camera is now regularly not available to my desktop. It works for about half an hour, then something seems to break and it will no longer connect. Restarting sometimes fixes the issue, but not usually. However, another weird detail I've noticed - the feeds work fine on my Android phone, at least most of the time. It seems like my Windows 10 PC specifically is having the issue. Maybe a session problem or something? I can't seem to figure out where the hang-up is. Tried different browsers, VLC, etc. Can still Ping the cameras.

bill-mcgonigle, 2022-05-17:
I can't explain this, but I wrote a script to restart `v4l2rtspserver` and log the output for debugging but with logging on the dang thing won't crash.  I mean, mixed blessings.  If anybody else wants to try:

`/root/restart-rtspserver`:
```
#!/bin/ash

PID=`pidof v4l2rtspserver`
if [ "$PID" = "" ]; then 
  mkdir -p /sdcard/debug
  date >> /sdcard/debug/rtspdead.log
  /usr/bin/v4l2rtspserver -v -A 48000 -C 1 -a S16_LE -l 1 /dev/video3,hw:0,1 /dev/video5 > /sdcard/debug/rtsp-`date -Iseconds`.log 2>&1  &
fi
```

in `crontab`:
```
* * * * * /root/restart-rtspserver >> /sdcard/debug/cron-rtspserver 2>&1
```

jjmuriel, 2022-05-25:
Same issue here. Maybe, there will be new firmware release with the solution for the RTSP Server crash?, @tachang?

EDPittore, 2022-06-09:
I been trying to resolve this exact issue. Numerous restarts of the camera didn't resolve anything. I could SSH into the camera, but none of the streams were working. Taking out the sdcard immediately resolved the issue. Scanning the sdcard some errors were encountered. What is strange is that the card was mounted successfully and accessible in the camera.  Testing to see if it's an issue with all sdcards or only those with unspecified "errors".

Nordikk, 2022-06-09:
My cam just stopped working completely. SSH works, but no video stream. Gonna reflash it

tachang, 2022-06-09:
I've noticed SSH tends to stay up but the other processes like `videocapture` tend to die.
I don't think reflashing it will help.

The sdcard is fat32 and is prone to issues so I have to use the fsck binary that I've compiled to fix it. Sometimes I will scp it over to the camera.

My other cameras have been pretty stable though overall.

EDPittore, 2022-06-09:
Results from some additional testing. It seems like if the camera is suddenly powered off there's a chance that the sdcard doesn't unmount cleanly. Next boot you get a message such as 

`openmiko kern.warn kernel: [    6.751223] FAT-fs (mmcblk0p1): Volume was not properly unmounted. Some data may be corrupt. Please run fsck.`

While this "corrupted" card is present the camera won't boot properly unless you either reboot without the card or perform some sort of repair. Even Windows Scan and Repair utility was sufficient.

My sdcards were all fat32 formatted.

Nordikk, 2022-06-09:
So I reflashed and it didn't work.
Then I ran Windows Scan and Repair on the sd card, works fine now.

bill-mcgonigle, 2022-06-09:
Corrupt sdcard may be a separate issue, though it doesn't make sense to not be able to repair a filesystem on these types of devices (they mostly run in unreliable power situations).  It looks like for large SD cards on embedded a low-memory branch of dosfstools is required: https://github.com/arnout/dosfstools/tree/reduce-memory-v2

bill-mcgonigle, 2022-06-09:
Guys, I filed #70 to talk about sdcard repair, having run into it with a power problem this morning.  Losing RTSP after hours to days (this bug) appears to be a separate cause of losing video.

Nordikk, 2022-06-11:
Since reparing the SD Card and reinstalling OpenMiko, my cam didn't crash once. Camera feed is as stable as it was never before

martinbutt, 2022-09-04:
I've attempted to address this with https://github.com/openmiko/openmiko/pull/81 and https://github.com/openmiko/openmiko/pull/82. Any extra eyes on those or testers would be greatly appreciated.

symphonymarie, 2022-09-17:
I actually ended up solving my own issue here by setting my MTU about 100 lower on the PC I was accessing the cams from. Not sure if this will work for everyone, but easy enough to try and revert if it doesn't. (Incidentally, this was my thought at the time, but then it actually worked!)