openmiko/openmiko #71: Great firmware! I have some primary questions

RS00001, 2022-06-10:
1. Like Defang,  does this solution require firmware to be on the SD card? Or does it install the firmware on the camera's embedded memory?
2. Is there any solution for discovery and pairing available? 
3. We simply want to get t20 cameras with their standard factory SDK, then replace it with this new firmware. 
4. Any plan to support t31?
Thanks again, team. I love the solution, so apologies if my questions are very basic.

tachang, 2022-06-11:
1. This solution does not need an SDcard. It can run entirely from the camera's embedded memory. That said there are certain features that rely on the sdcard. It can write logs to the sdcard and it can use the sdcard as swap memory.
2. Nope. Discovery is done via DHCP so you can check your router.
3. That *should* work but I haven't been able to get any Xiaomi Xiaofangs lately. I tried ordered on aliexpress but after 4 months I filed a compliant and got my money refunded. Wasn't sure if it was scam or not. If anyone has a hookup I'd like to get some.
4. Eventually would like to support T31 (wyzecam v3). I did some preliminary work and it seems to read the linux kernel off a hardcoded file name so you can use that to wipe the flash and rewrite.

Thanks for supporting,
Jeff

RS00001, 2022-06-12:
Thank you Jeff for your detailed reply.

1. Coudl you pleas suggest which features rely on SD card.
2. We are trying to achieve integration between hub and camera like arlo camera where you press a button on hub, keep camera around 3ft or less distance and then press button on camera too to make them pair. Will this normally be a part of the SDK? is it possible to achieve this customisation? 
3. Are there any supplier you think that would provide the camera with root password?
4. I am trying to find a t31 camera module with wifi but no luck so far

tachang, 2022-06-12:
1. The only feature that relies on the sdcard is the swapfile feature.
2. What do you mean by pair? OpenMiko doesn't really have a hub so there is nothing to pair with.
3. I think there are some providers on aliexpress.com
4. Have not found any 3rd party ones yet

RS00001, 2022-06-12:
1. thanks, its clear now
2. Hub and camera pairing, I know it is not available in OpenMico currently but we are looking to develop it and wanted to check if it's possible. 
3. thanks
4. same, no luck.

RS00001, 2022-06-12:
Btw, for the pairing option, are you (or anyone you know) open to consulting and building it for us on a commercial product - please let me know