openmiko/openmiko #72: Can't flash on iSmartAlarm Spot+

settecplus, 2022-06-14:
Hi,
I'm trying to flash Openmiko on the Spot+ camera by iSmartAlarm, but I can't seem to have any success...
I'm using a fat32 formatted SD card, with just the demo.bin file and the wpa_supplicant.conf file in /overlay/etc folder. 

For a starter, when booting with the reset button pressed the led never turns blue, it stays yellow. After about 20 seconds it starts blinking yellow but the cam won't connect to my wifi. I double checked the wpa_supplicant.conf file and it is correct (with Unix LF et all). If I keep pressing the reset button for more than 30 seconds I can hear a message stating "Please wait to connect", but nothing happens. 

After letting the camera run for a while I can see  a couple files and folders are created on the SD, but no log file. It seems the original OS is booting, not the new one. 

Did anybody had any success flashing Openmiko on this camera?

settecplus, 2022-06-15:
For those interested, I started a litle investigation. I connected via UART and it seems Root access with the well known password is working. 
I logged the boot process with Openmiko firmware renamed to FIRMWARE_660R.bin, with little luck:

```
U-Boot 2011.09 (Mar 18 2017 - 06:55:19)

DRAM:  64 MiB
MMC:   MMC: 0
SPI FLASH: 16 MB
In:    serial
Out:   serial
Err:   serial

Partition Map for MMC device 0  --   Partition Type: DOS

Partition     Start Sector     Num Sectors     Type
    1                 8192        62513152       c
reading FIRMWARE_660R.bin

11075648 bytes read
## Update firmware Starting  ...
ERROR: decrypt check failed
## Update firmware End, rc = 0xFFFFFFFF
ERROR: update failed

Partition Map for MMC device 0  --   Partition Type: DOS

Partition     Start Sector     Num Sectors     Type
    1                 8192        62513152       c
reading FIRMWARE_660R_F.bin

** Unable to read "FIRMWARE_660R_F.bin" from mmc 0:1 **
sd_update_fail: no FIRMWARE_660R_F.bin in the sd
ERROR: update FIRMWARE_F.bin from sd failed
Hit any key to stop autoboot:  0
roofsr size = 0x6e0070
## Booting kernel from Legacy Image at 00008000 ...
   Image Name:   Linux-2.6.35.12
   Image Type:   ARM Linux Kernel Image (uncompressed)
   Data Size:    3038112 Bytes = 2.9 MiB
   Load Address: 00008000
   Entry Point:  00008040
   Verifying Checksum ... OK
   XIP Kernel Image ... OK
OK

Starting kernel ...

Uncompressing Linux... done, booting the kernel.
```
So it seems it's a decryption error. 

The message:
`sd_update_fail: no FIRMWARE_660R_F.bin in the sd`

Gave me the idea to rename the file on the SD as FIRMWARE_660R_F.bin, and this time the boot did not return the "ERROR: decrypt check failed" message. But, unfortunately, it did nothing else, i.e. it seems I bricked the camera because the upgrade did not take place and the boot stopped.

tachang, 2022-06-25:
DRAM:  64 MiB

This means OpenMiko won't work here because the firmware is designed for 128MB devices. While it is possible to get it to fit in that size you would have to compiled a custom everything. Also this is ARM not MIPS.

tachang, 2022-06-26:
Closing this as not planned right now.

closed 2022-06-26 by tachang