openmiko/openmiko #73: Flashing doesn't connect to WiFi

jaymichaelic, 2022-07-08:
First of all, great work on this code.  I do really appreciate the support too.  Usually I can find the answers to my problems from reading other posts, which I have, but nothing seems to help solve my problem.

I recently purchased Waze Cam v2 for the sole purpose of using it for Octoprint.  I read all the documentation and had the SD card configured before it arrived!

Unfortunately, it's been a complete bust.  I know I put the files onto the SD card exactly how instructed.  I only changed the SSID and password and I made sure that the file extension was .conf rather than .conf.txt.  I also edited it in VS Code and made sure there were no extra spaces or funky characters.  Opened in notepad to be sure.  It is exactly as the example with only my SSID and password.

After holding down the setup button, plugging it in and waiting for a single blue flash before releasing the setup button.  It clicks a bit and then eventually starts flashing yellow.  I jump on my router's client table and see other devices but this one never arrives.  It just doesn't seem to want to connect to my network.

I have tried changing the security settings and have tested and retested with each new change and nothing works.  I have tried WPA-Personal, WPA - WPA2-Personal, WP2-Personal as well as changing it to AES, Both, and TKIP...all of the options.

I even tried using the 5GHz one even though I knew I should be using 2.4Ghz.  I tried everything under the sun on my router with no success.

I also tried changing the configuration file to remove the other items on the list so that it looked like this:

        key_mgmt=WPA-PSK
        pairwise=TKIP
        group=TKIP

Nothing seemed to do the trick.

What am I missing?

I have tried for hours, so if I can't get any additional help, I am just going to forget about having a webcam for my octoprint.

tachang, 2022-07-10:
@jaymichaelic - What micro sdcard are you using and what is it formatted as? It needs to be fat32 and I've heard of problems with big ones (larger than 16gb). I use 8gb and 16gb ones and it seems to work fine though. Are there any log files on the sdcard?

jaymichaelic, 2022-07-10:
Thank you for replying, Sir!

I thought about that too and kept it simple: 8GB formatted as Fat32.  No log files ever appear like some other posts.  In one post you mentioned that perhaps the bootloader was buggy, so I downloaded the stock bootloader as well but I couldn't figure out how to use it and I had already spent so much time that after the first try, I gave up.  After that I ended up just using it alongside my v3 cam.  I wonder if I tried flashing it now if it would work...maybe the bootloader already got updated with the firmware upgrade?  Haven't tried that yet.

tachang, 2022-07-11:
> After that I ended up just using it alongside my v3 cam.
Does that mean OpenMiko never actually flashed? Because if so it does sound like you never got the firmware onto the camera. The fact that there are no logs bolsters this assumption.

jaymichaelic, 2022-07-15:
What can I do if it will not flash onto the camera?

jdmz1, 2022-08-30:
I am getting logs but it seems to not be reading my wpa supplicant file at all:

wpa_supplicant v2.5
random: Trying to read entropy from /dev/random
Successfully initialized wpa_supplicant
Initializing interface 'wlan0' conf '/etc/wpa_supplicant.conf' driver 'nl80211' ctrl_interface 'N/A' bridge 'N/A'
Configuration file '/etc/wpa_supplicant.conf' -> '/etc/wpa_supplicant.conf'
Reading configuration file '/etc/wpa_supplicant.conf'
ctrl_interface='/var/run/wpa_supplicant'
ctrl_interface_group='0'
ap_scan=1
bgscan='simple:30:-45:300'
Priority group 2
   id=0 ssid='EXAMPLE_WIFI_NAME'
nl80211: Could not add multicast membership for vendor events: -2 (No such file or directory)
rfkill: Cannot open RFKILL control device
nl80211: RFKILL status not available

Supplicant file:
# This file is used to configure wifi
#
# For more configuration option please see:
# https://w1.fi/cgit/hostap/plain/wpa_supplicant/wpa_supplicant.conf

ctrl_interface=/var/run/wpa_supplicant
ctrl_interface_group=0
ap_scan=1
bgscan="simple:30:-45:300"

network={
        ssid="My network"
        # Uncomment to connect to Hidden SSIDs
        scan_ssid=1 
        key_mgmt=WPA-PSK