openmiko/openmiko #74: Dafang with v0.0.41-alpha.2 - no video stream

PeterRob, 2022-07-13:
v4l2rtspserver does not load on boot, I ran from the CL:

`# /usr/bin/v4l2rtspserver -A 48000 -C 1 -a S16_LE -l 1 /dev/video3,hw:0,1 /dev/video5
log level:500

[NOTICE] main.cpp:309
        Version: 1 live555 version:2016.01.29
[NOTICE] src/V4l2RTSPServer.cpp:36
        Create V4L2 Source.../dev/video3
[NOTICE] src/V4l2Device.cpp:133
        driver:v4l2 loopback capabilities:85008003 mandatory:4000001
[NOTICE] src/V4l2Device.cpp:135
        /dev/video3 support output
[NOTICE] src/V4l2Device.cpp:136
        /dev/video3 support capture
[NOTICE] src/V4l2Device.cpp:138
        /dev/video3 support read/write
[NOTICE] src/V4l2Device.cpp:139
        /dev/video3 support streaming
[NOTICE] src/V4l2Device.cpp:225
        /dev/video3:H264 size:1920x1080 bufferSize:8294400
[NOTICE] src/V4l2Device.cpp:246
        fps:1/25
[NOTICE] src/V4l2Device.cpp:247
        nbBuffer:2
[NOTICE] src/V4l2MmapDevice.cpp:49
        Device /dev/video3
[NOTICE] src/V4l2MmapDevice.cpp:73
        Device /dev/video3 nb buffer:2
[NOTICE] src/V4l2MmapDevice.cpp:49
        VIDIOC_REQBUFS: Inappropriate ioctl for device
Device video3
[NOTICE] src/V4l2MmapDevice.cpp:141
        VIDIOC_STREAMOFF: Inappropriate ioctl for device
VIDIOC_REQBUFS: Inappropriate ioctl for device
Device video3
[WARN] src/V4l2RTSPServer.cpp:52
        Cannot open output:video3
[NOTICE] src/V4l2RTSPServer.cpp:61
        Create Source .../dev/video3
[NOTICE] src/AudioCompressor.cpp:16
        Initializing AudioCompressor
[NOTICE] src/MP3AudioCompressor.cpp:67
        Initializing MP3AudioCompressor
[NOTICE] src/DeviceSource.cpp:93
        begin thread
[NOTICE] src/MP3AudioCompressor.cpp:35
        LAME 3.99.5 32bits (http://lame.sf.net)

[NOTICE] src/MP3AudioCompressor.cpp:35
        Using polyphase lowpass filter, transition band: 16452 Hz - 17032 Hz

[NOTICE] main.cpp:373
        Using AudioCompressor: 0x53de88
[NOTICE] src/V4l2RTSPServer.cpp:183
        Create ALSA Source...hw:0,1
[NOTICE] src/ALSACapture.cpp:21
        params.m_audioCompressor: 0x53de88
[NOTICE] src/ALSACapture.cpp:54
        Open ALSA device: "hw:0,1"
[NOTICE] src/ALSACapture.cpp:132
        set sample format device: hw:0,1 to:2 ok
[NOTICE] src/ALSACapture.cpp:109
        ALSA device: "hw:0,1" buffer_size:1048576 period_size:1024 rate:48000
[NOTICE] inc/ServerMediaSubsession.h:49
        format:video/H264
[NOTICE] inc/ServerMediaSubsession.h:49
        format:audio/L16/48000/1
[NOTICE] src/UnicastServerMediaSubsession.cpp:35
        Compressed audio format:audio/MPEG
[NOTICE] inc/V4l2RTSPServer.h:82
        Play this stream using the URL "rtsp://192.168.131.15:8554/video3_unicast"
[NOTICE] src/V4l2RTSPServer.cpp:36
        Create V4L2 Source.../dev/video5
[NOTICE] src/V4l2MmapDevice.cpp:49
        VIDIOC_REQBUFS: Inappropriate ioctl for device
Device /dev/video5
[NOTICE] src/V4l2MmapDevice.cpp:141
        VIDIOC_STREAMOFF: Inappropriate ioctl for device
VIDIOC_REQBUFS: Inappropriate ioctl for device
Device /dev/video5
[NOTICE] src/AudioCompressor.cpp:16
        Initializing AudioCompressor
[NOTICE] src/MP3AudioCompressor.cpp:67
        Initializing MP3AudioCompressor
[NOTICE] src/DeviceSource.cpp:93
        begin thread
[NOTICE] src/MP3AudioCompressor.cpp:35
        LAME 3.99.5 32bits (http://lame.sf.net)

[NOTICE] src/MP3AudioCompressor.cpp:35
        Using polyphase lowpass filter, transition band: 16452 Hz - 17032 Hz

[NOTICE] main.cpp:373
`
The RTSP stream is available but not http

Also time does not synchronise with the NTP server.

tachang, 2022-07-14:
Did you update openmiko.conf if you were upgrading from older version?

tachang, 2022-07-14:
It is weird it is not loading on boot. If you have any logs to provide that would be great.

PeterRob, 2022-07-14:
I reflashed the alpha, now time is synciing with my NTP server. messages, videocapture.log & openmiko.conf are here - [https://gofile.io/d/UCoZal](url)

bill-mcgonigle, 2022-11-27:
I had this after flashing 3e9737c .  I put a copy of openmiko.conf from master in the overlay and rebooted and then it worked.  It looks like the VIDEO_DEV_ lines are the difference.

PeterRob, 2022-11-28:
> I had this after flashing [3e9737c](https://github.com/openmiko/openmiko/commit/3e9737c0d8406092c1aa7a913ab7b0568d308813) . I put a copy of openmiko.conf from master in the overlay and rebooted and then it worked. It looks like the VIDEO_DEV_ lines are the difference.

Thanks for the heads-up, with those lines it worked, I must have overlooked tachang's question on updating openmiko.conf