openmiko/openmiko #75: no wifi Wyze Cam V2

mon5termatt, 2022-07-13:
```wpa_supplicant v2.5
random: Trying to read entropy from /dev/random
Successfully initialized wpa_supplicant
Initializing interface 'wlan0' conf '/etc/wpa_supplicant.conf' driver 'nl80211' ctrl_interface 'N/A' bridge 'N/A'
Configuration file '/etc/wpa_supplicant.conf' -> '/etc/wpa_supplicant.conf'
Reading configuration file '/etc/wpa_supplicant.conf'
ctrl_interface='/var/run/wpa_supplicant'
ctrl_interface_group='0'
ap_scan=1
Priority group 2
   id=0 ssid='Get off my LAN'
nl80211: Could not add multicast membership for vendor events: -2 (No such file or directory)
rfkill: Cannot open RFKILL control device
nl80211: RFKILL status not available
Could not read interface wlan0 flags: No such device
nl80211: Driver does not support authentication/association or connect commands
nl80211: deinit ifname=wlan0 disabled_11b_rates=0
nl80211: Remove monitor interface: refcount=0
netlink: Operstate: ifindex=0 linkmode=0 (kernel-control), operstate=6 (IF_OPER_UP)
Could not read interface wlan0 flags: No such device
nl80211: Set mode ifindex 0 iftype 2 (STATION)
nl80211: Failed to set interface 0 to mode 2: -19 (No such device)
wlan0: Failed to initialize driver interface
Failed to add interface wlan0
wlan0: Cancelling scan request
wlan0: Cancelling authentication timeout```

SSID and password is defiantly correct

tachang, 2022-07-14:
Is `key_mgmt=WPA-PSK` set?

LilTrublMakr, 2022-07-31:
I am getting the same. 

Logs:
```
wpa_supplicant v2.5
random: Trying to read entropy from /dev/random
Successfully initialized wpa_supplicant
Initializing interface 'wlan0' conf '/etc/wpa_supplicant.conf' driver 'nl80211' ctrl_interface 'N/A' bridge 'N/A'
Configuration file '/etc/wpa_supplicant.conf' -> '/etc/wpa_supplicant.conf'
Reading configuration file '/etc/wpa_supplicant.conf'
ctrl_interface='/var/run/wpa_supplicant'
ctrl_interface_group='0'
ap_scan=1
bgscan='simple:30:-45:300'
Priority group 2
   id=0 ssid='MINE-IoT'
nl80211: Could not add multicast membership for vendor events: -2 (No such file or directory)
rfkill: Cannot open RFKILL control device
nl80211: RFKILL status not available
Could not read interface wlan0 flags: No such device
nl80211: Driver does not support authentication/association or connect commands
nl80211: deinit ifname=wlan0 disabled_11b_rates=0
nl80211: Remove monitor interface: refcount=0
netlink: Operstate: ifindex=0 linkmode=0 (kernel-control), operstate=6 (IF_OPER_UP)
Could not read interface wlan0 flags: No such device
nl80211: Set mode ifindex 0 iftype 2 (STATION)
nl80211: Failed to set interface 0 to mode 2: -19 (No such device)
wlan0: Failed to initialize driver interface
Failed to add interface wlan0
wlan0: Cancelling scan request
wlan0: Cancelling authentication timeout
```
WPA Supplicant:
```
ctrl_interface=/var/run/wpa_supplicant
ctrl_interface_group=0
ap_scan=1
bgscan="simple:30:-45:300"

network={
        ssid="MINE-IoT"
        key_mgmt=WPA-PSK
        pairwise=CCMP TKIP
        group=CCMP TKIP WEP104 WEP40
        psk="DefinitelyNotMyPassword"
        priority=2
}
```
This is using the latest release on a cam v2.

mon5termatt, 2022-08-01:
I forgot to mention I was using the latest version of v2 as well

larrywright, 2022-08-12:
I’m getting this same issue with a Wyze Pan Cam. I tried the latest version as well as several older versions. Everything seems to flash fine, and I see logs on the SD card, but the Wifi never connects and I see the “no such device” error mentioned above.

eviled666, 2022-09-02:
getting this as well

eviled666, 2022-09-02:
getting this same error

mmarbut, 2022-09-14:
same issue with my wyze cam pan
```

wpa_supplicant v2.5
random: Trying to read entropy from /dev/random
Successfully initialized wpa_supplicant
Initializing interface 'wlan0' conf '/etc/wpa_supplicant.conf' driver 'nl80211' ctrl_interface 'N/A' bridge 'N/A'
Configuration file '/etc/wpa_supplicant.conf' -> '/etc/wpa_supplicant.conf'
Reading configuration file '/etc/wpa_supplicant.conf'
Priority group 2
   id=0 ssid='MrUniverse Cortex Uplink'
nl80211: Could not add multicast membership for vendor events: -2 (No such file or directory)
rfkill: Cannot open RFKILL control device
nl80211: RFKILL status not available
Could not read interface wlan0 flags: No such device
nl80211: Driver does not support authentication/association or connect commands
nl80211: deinit ifname=wlan0 disabled_11b_rates=0
nl80211: Remove monitor interface: refcount=0
netlink: Operstate: ifindex=0 linkmode=0 (kernel-control), operstate=6 (IF_OPER_UP)
Could not read interface wlan0 flags: No such device
nl80211: Set mode ifindex 0 iftype 2 (STATION)
nl80211: Failed to set interface 0 to mode 2: -19 (No such device)
wlan0: Failed to initialize driver interface
Failed to add interface wlan0
wlan0: Cancelling scan request
wlan0: Cancelling authentication timeout
```

Tukamok, 2022-09-20:
I had the same issues.  Cam would get on the network using the stock firmware, so I knew it wasn't a hardware issue.  I'm relatively certain one of the two below solved the issue:

1.  Re-Flashed several times, just to be safe.
2. Ensured that my wpa_supplicant.conf was using unix, and not windows, line endings.

Yes, I know that the docs said that #2 shouldn't be an issue with the latest, but I was running out of things to try before I gave up.

mmarbut, 2022-09-20:
Got the answer on the slack at the bottom of this you have to change the wifi module.
https://github.com/openmiko/openmiko/blob/master/overlay_minimal/etc/openmiko.conf
then put the file in the same folder as the supplicant folder.

mon5termatt, 2022-09-20:
> Got the answer on the slack at the bottom of this you have to change the wifi module. https://github.com/openmiko/openmiko/blob/master/overlay_minimal/etc/openmiko.conf then put the file in the same folder as the supplicant file.

is that for the pan or v2?

mmarbut, 2022-09-20:
the pan, but it would be worth doublechecking and making sure that you have the right module.

LilTrublMakr, 2022-09-20:
> Got the answer on the slack at the bottom of this you have to change the wifi module. https://github.com/openmiko/openmiko/blob/master/overlay_minimal/etc/openmiko.conf then put the file in the same folder as the supplicant file.

Not relevant for the V2. We are having problems with the V2 specifically.
@tachang any input on this?

larrywright, 2022-09-21:
I can confirm that the issue I was having was related to the wifi module. Changing to the appropriate one resolved it.

donwade, 2022-09-24:
I instrumented events.c in the wpa component. It DOES NOT seem to be picking up the wpa_supplicant.conf file FROM THE SD card to get the desired SSID and password from wpa_supplicant.conf, but always the one found in github under .../overlay_minimal....
So  until one figures out where in the code/scripts it copies out the SD card data and overstrikes/overlays the internal wpa_supplicant.conf file , you're going to have to edit wpa_supplicant.wpa (under the overlay_minimal directory) and issue a make. Then copy out the demo.bin (which will then have your hard embedded config in it) and put it on the SD card.
Given the poor workflow provided on this project, or everyone here is a docker-jedi, the exercise of getting a docker container console, copying docker files to a external SD card is left up to the student :(.
I believe people who have this issue and only use the prebuilt binaries (and don't know  or want to know about docker/containers/make) are left behind.

martinbutt, 2023-02-20:
I decided to merge together all of my changes from the last six months and [release them for testing](https://github.com/martinbutt/openmiko/releases/tag/v1.0.0-alpha.1). I've included new config file generators that ask which camera you are using and outputs appropriate config files. Testing is appreciated, and you can [file an issue](https://github.com/martinbutt/openmiko/issues) in that repo if you have any problems.

mon5termatt, 2023-02-20:
Sounds like a plan. I still have one camera out of three that's just refusing to work with it so I'll give it a try tomorrow

martinbutt, 2023-02-20:
Great, thanks! To get the camera and SD card back to a clean slate before starting, you can:
- Hold the reset button for a few seconds and it will flash blue three times to show it has deleted any residual files
- Format the memory card, as it can easily become corrupted if it is writing to a log file when it is powered off.

closed 2024-03-18 by mon5termatt