openmiko/openmiko #76: SD card size matters

reidprichard, 2022-07-13:
Wyze Cam v2

I couldn't find this mentioned anywhere but a random Reddit thread (after hours of pulling my hair out and fearing I'd bricked my device), but it seems that the size of the SD card you use is crucial. Nothing I could try would get firmware to flash with a 16GB card (formatted FAT32 as instructed), but flashing immediately worked on an old 128MB card I had (also FAT32).

tachang, 2022-07-14:
Thank you for reporting this. I had my suspicions something was up. I'm surprised a 16GB card formatted as FAT32 didn't flash. I have 16GB cards as well. What brand was it?

jschwalbe, 2022-07-14:
More likely is that it’s a bad SD card.

reidprichard, 2022-07-14:
It's a SanDisk Ultra class 10 microSD. 

I've never used it for anything else, and it is of unknown provenance, so it's certainly possible that it is bad. In that case, though, would my PC still be able to format/read/write it?

reidprichard, 2022-07-14:
If that's the case, perhaps it would be worth adding a cautionary note to that effect to the readme?

donwade, 2022-09-25:
Perhaps pop out the SD card and post /var/log/messages and /var/log/wpa_supplicant.log?