openmiko/openmiko #78: How to disable WiFi and only use USB Ethernet?

diogosalazar, 2022-07-28:
As the title says, I would like to keep WiFi permanently off and have the camera stream over Ethernet only.
Thanks

MartyTemple, 2022-08-28:
Having an option for a hard wired connection would be great but, I believe the webcam directly over USB would be better (higher priority.) I am using the Printoid app and  I would really not like to expose my OpenMiko/Wyze camera to the internet directly. Enabling the webcam USB port would allow the Raspberry Pi to see the video stream and allow it to be shared the external Octoprint plugins (Printoid.)

psychosb, 2022-08-28:
> Having an option for a hard wired connection would be great but, I believe the webcam directly over USB would be better (higher priority.) I am using the Printoid app and  I would really not like to [...]

This would be awesome.

niharmehta, 2023-06-03:
Bumping this one back up.  
Camera is in a location without good wifi or a RPI, but I do have ethernet.  Looking at Dafang, activating the usb-ethernet option is as simple as dropping a config file in the config directory and the run.sh will detect this and load the usbnet and asix drivers as well as bring up eth0.  

I can do this manually after boot however, I am looking to find the equivalent automatic method at boot in OpenMiko as Dafang as it seams like OpenMiko does not use the same run.sh process. 
What would be the best process for this?  Create a new start script in the /config/overlay/etc/rc.d  which loads the drivers or is there a better, more supported way to do it ?

exaiyo, 2023-10-18:
> Bumping this one back up. Camera is in a location without good wifi or a RPI, but I do have ethernet. Looking at Dafang, activating the usb-ethernet option is as simple as dropping a config file in [...]

Were you able to figure out how to do this? I didn't realize this functionality was already possible, albeit unofficially.