openmiko/openmiko #79: How to enable RTSP?

dominikheinz, 2022-08-16:
Hey,
I flashed openmiko, and got mostly everything working, including the stream via HTTP.
However, it doesnt seem that RTSP is working (The documentation mentions rtsp://Your_Camera_IP:8554/video3_unicast), however, after performing a portscan, I noticed that this port is not even open on the camera.
Do I have to enable or start the RTSP server somehow?

martinbutt, 2022-09-22:
RTSP is a UDP port, not TCP. Scanning UDP ports is [difficult by their nature](https://nmap.org/book/scan-methods-udp-scan.html), so typically disabled by default on port scanners.

To test the port I recommend using [VLC](https://www.videolan.org/vlc/). Open VLC > Media > Open Network Stream > Enter cam URL (rtsp://Your_Camera_IP:8554/video3_unicast) > Play

If you need authentication on the endpoints, see: https://github.com/openmiko/openmiko/blob/master/overlay_minimal/etc/v4l2rtspserver.conf.example

martinbutt, 2022-10-31:
You are right. I had misread RTP as RTSP when looking this up myself. Port 8554 doesn't appear for me when I port scan my cameras, but the stream does work. I'm not sure why then.

As for your streams only coming up half the time, is there anything in /var/log/ that looks like it could be an issue? Also, which version are you using?

dominikheinz, 2022-11-07:
> RTSP is generally TCP but I think it can be done over UDP in some cases.
> 
> There seems to be around a 50:50 success rate with RTSP activating, I have to reboot until it comes back on. When RTSP doesn't load, SSH/HTTP still works.

Yea, exactly this. Sometimes RTSP just doesn't get activated. Maybe it would make sense to have a watchdog script running on the device that checks if the service is running or not, and starts it if necessary. (That could also be helpful in case of a crash of the server etc)

sco-gar, 2022-11-07:
> Yea, exactly this. Sometimes RTSP just doesn't get activated. Maybe it would make sense to have a watchdog script running on the device that checks if the service is running or not, and starts it if [...]

I think this does happen in some cases (RTSP not activating) but I was being a bit stupid - I was rushing and made an error with the order of my firewall rules, I kept adjusting the rules while trying to get RTSP running stable. In the end, I think I now have a somewhat stable RTSP feed from 2x Neo's/Wyze v2.

I used openmiko.conf (in /config/overlay/etc) - initially I uncommented each setting to disable the features I did not want (eg. ENABLE_API=0, ENABLE_MJPEG_OVER_HTTP=0, ENABLE_AUTONIGHT=0) - I then rebooted to apply.

I then commented out everything except VIDEO_DEV_1/2 and WIFI_MODULE - this seems to have retained the previous settings as the HTTP stream/autonight etc are all still disabled. HTTP disabled to save overhead, no autonight as I'm using it against a window. I disabled VIDEO_DEV_3 because I don't use a sub-stream currently but will re-able if needed.

So far this has given me a steady RTSP stream, with SSH and no HTTP. Here is my current openmiko.conf (and you can see what I previously disabled/set to =0):

`# OpenMiko configuration file

# ENABLE_DEBUG=0

# Starts ffmpeg on startup recording from the v4l2rtspserver
# ENABLE_RECORDING=0

# Start the micropython based api
# ENABLE_API=0

# Use a swapfile on the sdcard
# ENABLE_SWAP=1
# Use this setting if swapfile is a dedicated partition
# SWAPFILE=/dev/mmcblk0p1

# Set parition to use for mounting at /sdcard
# SD_PARTITION=/dev/mmcblk0p2
# SD_FILESYSTEM=vfat

# Enable MJPEG over HTTP
# ENABLE_MJPEG_OVER_HTTP=0
# MJPEG_FPS=1

# Script that detects nighttime/daytime and turns on IR LEDs
# ENABLE_AUTONIGHT=0
# AUTONIGHT_PARAMS="-j 3 -w 3 -1 1200000 -2 930000,14,10 -3 3000,17,8"

# Set alternate videocapture settings file
# VIDEO_CAPTURE_SETTINGS=/etc/videocapture_settings_1_encoder.json

# Set enable audio (is currently buggy, use at your own risk!)
# ENABLE_AUDIO=0

# Make sure these settings match the /etc/videocapture_settings.json file
VIDEO_DEV_1=/dev/video3
VIDEO_DEV_2=/dev/video4
# VIDEO_DEV_3=/dev/video5

# Use 8189fs for WyzeCam V2. For the WyzeCam Pan and Dafang use 8189es
WIFI_MODULE=8189fs

# Disables LEDs on the camera
# DISABLE_LEDS=0`