openmiko/openmiko #88: wpa_supplicant fails to initialize wlan0

Kendrik01, 2022-11-05:
Version: 0.0.40 / 0.0.41
Device: Wyze Cam V2

Issue: I am attempting to re-flash a camera that was already running an OLD version of OpenMiko.  I've re-flashed it at least 10 times with various versions and I keep seeing the same issue in the wpa_supplicant.log saying it can't initialize wlan0.  I've attached the wpa_supplicant.log below.

**wpa_supplicant.conf:**

```
# This file is used to configure wifi
#
# For more configuration option please see:
# https://w1.fi/cgit/hostap/plain/wpa_supplicant/wpa_supplicant.conf

ctrl_interface=/var/run/wpa_supplicant
ctrl_interface_group=0
ap_scan=1
bgscan="simple:30:-45:300"

network={
        ssid="Oyler-Home"
        # Uncomment to connect to Hidden SSIDs
        #scan_ssid=1 
		proto=RSN
        key_mgmt=WPA-PSK
        pairwise=CCMP TKIP
        group=CCMP TKIP WEP104 WEP40
		# psk stands for pre-shared key (the wifi password)
        psk="my_password"
        priority=2
}
```

**openmiko.conf:**
```
# OpenMiko configuration file

#ENABLE_DEBUG=0

# Starts ffmpeg on startup recording from the v4l2rtspserver
#ENABLE_RECORDING=0

# Start the micropython based api
ENABLE_API=1

# Use a swapfile on the sdcard
# ENABLE_SWAP=1
# Use this setting if swapfile is a dedicated partition
# SWAPFILE=/dev/mmcblk0p1

# Set parition to use for mounting at /sdcard
# SD_PARTITION=/dev/mmcblk0p2
# SD_FILESYSTEM=vfat

# Enable MJPEG over HTTP
ENABLE_MJPEG_OVER_HTTP=1
MJPEG_FPS=15

# Script that detects nighttime/daytime and turns on IR LEDs
ENABLE_AUTONIGHT=1
AUTONIGHT_PARAMS="-j 3 -w 3 -1 1200000 -2 930000,14,10 -3 3000,17,8"

# Set alternate videocapture settings file
# VIDEO_CAPTURE_SETTINGS=/etc/videocapture_settings_1_encoder.json

# Set enable audio (is currently buggy, use at your own risk!)
# ENABLE_AUDIO=0

# Make sure these settings match the /etc/videocapture_settings.json file
VIDEO_DEV_1=/dev/video3
VIDEO_DEV_2=/dev/video4
VIDEO_DEV_3=/dev/video5

# Use 8189fs for WyzeCam V2. For the WyzeCam Pan and Dafang use 8189es
WIFI_MODULE=8189fs

# Disables LEDs on the camera
DISABLE_LEDS=0
```

[wpa_supplicant.log](https://github.com/openmiko/openmiko/files/9942676/wpa_supplicant.log)

martinbutt, 2022-11-05:
Make sure you are using fresh config files, as the format has changed with each release. 

If that doesn't solve it, to ensure you have cleared any residual files from the camera, hold the reset button for a few seconds and it will flash blue three times. You should also format the memory card, as it can easily become corrupted if it is writing to a log file when it is powered off.

Kendrik01, 2022-11-05:
> Make sure you are using fresh config files, as the format has changed with each release.
>
> If that doesn't solve it, to ensure you have cleared any residual files from the camera, hold the reset [...]

It's a brand-new configuration file.  I've also reset the camera and the memory card has been formatted.  Twice.

martinbutt, 2022-11-05:
The openmiko.conf file you have will only work with 0.41, so stick with that. I realize you have tried with that already, but just to narrow things down. Also make sure you have 0.41-alpha2.

Are there any other files in your config directory on the sdcard?

I've noticed you have `proto=RSN` in your wpa_supplicant.conf. Just because it is not in the original file could you try without it? I don't see any reason removing it would help, but I've had issues with seemingly minor differences before.

I was also finding that the memory card was corrupting every time I turned off the camera. I have a separate .bin file with a handful of changes on it (including fsck) that runs nicely for me on a WyzeCam v2. If all else fails, I can try sending that.

Kendrik01, 2022-11-05:
Okay,  I went ahead and re-formatted the SD card then made fresh copies of openmiko.conf, wpa_supplicant.conf, and videocapture_settings.json and dropped them into /config/overlay/etc .  I also downloaded a fresh copy of the 0.0.41-alpha2 and moved that to the SD card and renamed it to demo.bin.  And...it worked.  No idea what the issue was (I swear I've TRIED this exact configuration before) but hey, 37th times the charm!  Thank you for the assistance.

closed 2022-11-05 by Kendrik01

martinbutt, 2022-11-05:
I'm glad that worked. My money is on a corrupted SD card, and/or a slightly
wrong config file. Both have caught me out before. A future version should
have fsck built in. I'm just waiting on feedback from the maintainer on my
approach of moving to optional modules. I can also work on a config file
linter to catch inconsistencies in the config files.

martinbutt, 2022-11-05:
I've added a PR with some basic config generators: https://github.com/openmiko/openmiko/pull/89

Going down the linting route was going to be a lot harder, and this should cover most people's use cases.