openmiko/openmiko #92: Can I use this with an Ingenic T21 camera?

felipecrs, 2022-11-25:
Specifically this one: http://www.howellcam.com/product/38.html

CDX-24, 2022-12-14:
Hello there.

I also have several cameras with Ingenic T21 and ov2735b as the sensor (Wansview brand).
I am currently trying to alter the firmware but I think I will try with openmiko.
Also interested if some have experience and knowledge to share with this hardware !

Regards.

martinbutt, 2022-12-15:
I have no specific experience, but if you follow these [updated instructions](https://github.com/openmiko/openmiko/blob/963f054ec55c0ebd92992cdb9e885194e7a5ad7b/doc/development.md), you'll have a Docker Buildroot environment to work in.

The code for the camera sensor is in [this repo](https://github.com/openmiko/ingenic_videocap), but is not part of the project I'm very familiar with.

In [this fork](https://github.com/q09029/ingenic_videocap) @q09029 was working on adding support for the T31. I don't know how far they got, but their commits might help you along.

CDX-24, 2022-12-16:
Thank you I'll be sure to check this out ! 😄

ryanjv755, 2022-12-16:
Hey CDX-24,

My friend and I are currently digging into this one. Let me know if we can help.

gtxaspec, 2024-01-07:
Ingenic T21 and ov2735 now supported in openipc. FYI

closed 2024-01-07 by felipecrs