openmiko/openmiko #94: No video stream from camera, Failed to fetch data: no such table: openmiko_cameraprofile

blathers123, 2022-12-20:
I just tried flashing the latest alpha (v0.0.41-alpha.2) to my v2 and my pan. I'm not able to get any video feed and in the videostream logs I see:

`ERROR /openmiko/build/buildroot-2016.02/output/build/ingenic_videocap-86c1576153c6897f0c3e5b350a9f963363d69161/src/capture.c:838: Failed to fetch data: no such table: openmiko_cameraprofile`

martinbutt, 2022-12-26:
I've not seen this issue before. Did you try reformatting the SD card (helps is the filesystem is corrupt), and reflashing?

blathers123, 2022-12-26:
I didn't try reformatting the SD card. 
I sshed into the camera, deleted the database. Copied the camera profile from the config partition to the SD card, and reflashed the camera a few times. Eventually it worked. Not sure exactly what the issue was or which one of those things (if any) fixed it

blathers123, 2022-12-26:
Sorry, fat fingered the close button

martinbutt, 2023-02-20:
I decided to merge together all of my changes from the last six months and [release them for testing](https://github.com/martinbutt/openmiko/releases/tag/v1.0.0-alpha.1). This should address the issue that it stops working after a while. Testing is appreciated, and you can [file an issue](https://github.com/martinbutt/openmiko/issues) in that repo if you have any problems.