openmiko/openmiko #95: Audio is not working, also modified configuration to provide two rtsp streams

onkar201988, 2022-12-21:
I tweaked the settings to get two different streams, one with 1080P 25FPS and another one with 360P 10FPS. I am planning to use these with Frigate in Home Assistant.
But whatever I tried, I am not able to get the audio working. I am attaching my overlay files here. I also tried the default overlay files, still no luck.

Please help me to get the audio working. 
All in all great firmware 👍 
[Overlay.zip](https://github.com/openmiko/openmiko/files/10282256/Overlay.zip)

martinbutt, 2022-12-26:
For the audio, it's currently an experimental implementation. Which camera are you using?

For the IR LEDs, do you have the API enabled? If not, it will not work in the current version. There is an open PR to fix this.

onkar201988, 2022-12-26:
I am using Wyze cam 2. 
I am not sure which API to enable. I will wait till PR is merged.

martinbutt, 2022-12-28:
I can't hep with the audio until I have some free time to do a deep dive into the implementation.

For the IR LEDs, in your openmiko.conf, ensure `ENABLE_API=1` is set.

onkar201988, 2023-01-03:
Thank you for IR LED help. Can you point where I can check for audio in the code? I will try to check from my side.

martinbutt, 2023-02-20:
@onkar201988 the audio is in a separate repo here: https://github.com/openmiko/ingenic_videocap/blob/bdc7de80cc9eeee45ce5f16383289f8d663c838a/src/capture.c

If you want to devlop that app, the easiest way is using the developer instructions here: https://github.com/martinbutt/openmiko/blob/master/doc/development.md

bikemike, 2023-12-29:
@onkar201988 Thanks for the dual rtsp streams.

I was able to get audio working for both streams by doing the following:

Added/changed the following in openmiko.conf:
```
ENABLE_AUDIO=1

# Set which stream(s) will receive the audio
ENABLE_AUDIO_1=1
ENABLE_AUDIO_2=1
ENABLE_AUDIO_3=0
```

Added an overlay for /etc/init.d/S60camera and changed the following 3 things:

Changed the ffmpeg audio command:
```
ffmpeg -hide_banner -re -nostats -nostdin -f oss -sample_rate 8000 -channels 1 -i /dev/dsp -codec:a copy -f alsa hw:0,0 -codec:a copy -f alsa hw:0,1 > /dev/null 2>&1 &
```

Changed the way audio devices are appended to the rtsp streams (this won't work for 3 audio streams or if you want audio on VIDEO_DEV_2 and VIDEO_DEV_3):
```
                # Loop through devices to see if they're enabled and add them to rtspserver options
                if [ ! -z "$VIDEO_DEV_1" ]; then
                        DEVICES="${VIDEO_DEV_1}"
                        if [ "$ENABLE_AUDIO_1" == 1 ]; then
                                DEVICES="${DEVICES},hw:0,0"
                        fi
                fi
                if [ ! -z "$VIDEO_DEV_2" ]; then
                        DEVICES="${DEVICES} ${VIDEO_DEV_2}"
                        if [ "$ENABLE_AUDIO_2" == 1 ]; then
                                DEVICES="${DEVICES},hw:0,1"
                        fi
                fi
                if [ ! -z "$VIDEO_DEV_3" ]; then
                        DEVICES="${DEVICES} ${VIDEO_DEV_3}"
                        if [ "$ENABLE_AUDIO_3" == 1 ]; then
                                DEVICES="${DEVICES},hw:0,1"
                        fi
                fi
```

And finally, changed the the v4l2rtspserver command:
```
/usr/bin/v4l2rtspserver $AUTH_PARAMS -A 8000 -C 1 -a S16_BE -l 1 $DEVICES > /dev/null & #> /var/log/v4l2rtspserver.log 2>&1 &
```

I was able to get audio from this stream to work in Frigate too but that was much more of a headache!

onkar201988, 2024-01-11:
@bikemike thank you for providing information about the audio.

But I am facing some problem, I am not expert in linux. When I edit  file /etc/init.d/S60camera using nano and save it, I can see the changes are preserved for the session. But since I was not able to get the audio I restarted the camera and the file contents went back to default.

I tried to copy the file like how we can copy config files by keeping them in the microsd card, but that did not work as well.

How do you make changes and keep them forewer?

bikemike, 2024-01-13:
@onkar201988 I think the SD card files are under the /sdcard folder(I'm not at home at the moment so I can't verify). You need to edit files in there too have them stick. You can copy the S60camera file from /etc/init.d/ to the overlay/etc/init.d subfolder of the SD card.

onkar201988, 2024-01-14:
Thank you @bikemike , it is working for me as well now 👍

onkar201988, 2024-01-14:
As per comment from @martinbutt the IR cut filter is working and per comment from @bikemike the audio is working too. Thank you so much guys. Closing the issue.

closed 2024-01-14 by onkar201988