openmiko/openmiko #97: Problem with the load of files, from SDCARD /config/overlay/etc

xinux87, 2022-12-25:
Hi 

i've been trying to load the files from the last Version [Release v0.0.41-alpha.2] and i'm not able to see why isn't working, 

there is any log on the messages? if there is any problem coping the files?

used route: E:\config\overlay\etc

[config.zip](https://github.com/openmiko/openmiko/files/10299807/config.zip)

Thanks!!

P.D Merry Christmas!!

xinux87, 2022-12-25:
Even tried with linux, and it dont copy the config files, there is any change in the last version not shown in the wiki?

thanks

martinbutt, 2022-12-26:
The config files look fine to me. I've experienced this when the filesystem on the SD card has become corrupt. It can be fixed with fsck, but may just be easier to reformat. Make sure you use FAT32 and not exFAT, as that is not currently supported. There is PR to fix this that will hopefully be included in a future version.

Kaan88, 2022-12-26:
Facing the same issue here, did everything I could think of and looked through the issues. Reformatted many times and even booted up into linux just to be sure. 

My config files and folder structure is the same as above.

xinux87, 2022-12-26:
I've found a message on the messages file, saying that you've problems with a bad SDcard, please use fsck, but when the message it doesn't appear, it dont charge the files :S

Kaan88, 2022-12-26:
In my case according to wpa_supplicant.log, config for wifi gets copied, because otherwise it's reported as missing on the log file. 

So my real issue is that the wifi driver won't change from 8189fs to 8189es, maybe op's situation is the same.

jangliss, 2022-12-31:
Had the same issue, when I looked over the logs the only error about mounting the SD card was due to an unclean unmount. That's expected when you just unplug the power supply.  I ran through the process of reflashing it again without changing anything, and second time it worked.. Now I just have other issues.

martinbutt, 2023-02-20:
I decided to merge together all of my changes from the last six months and [release them for testing](https://github.com/martinbutt/openmiko/releases/tag/v1.0.0-alpha.1). There is a new option to disable logging in the config file. This should address the issue of the SD card filesystem becoming corrupted. It's also possible to load an fsck module to run on boot. Testing is appreciated, and you can [file an issue](https://github.com/martinbutt/openmiko/issues) in that repo if you have any problems.

claud9999, 2023-05-15:
I'm seeing this problem; both with 0.0.41a2 and 1.00a1. My config files are installed in /config/overlay/etc but the camera does not join the wifi as expected. The only way I could get this to work was to use the 0.0.40 release. Thoughts?

martinbutt, 2023-05-15:
@claud9999 did you try running fsck against the SD card? Could you also attach your config files just incase there is a parsing error that was introduced in the newer versions?

claud9999, 2023-05-15:
Spoke too soon on 0.0.40, FYI; it joined the wifi but never got to the state of running the RTSP server or web or ssh servers.

I've tried with two Wyze V2's and three different SD cards, all freshly formatted, and same behavior. With 0.0.41 and 1.0, no wifi even.

The configuration files as contained in <sd>/config/overlay/etc: 

[config.zip](https://github.com/openmiko/openmiko/files/11481239/config.zip)

claud9999, 2023-05-15:
also note both of these cams and the SD cards have been in service for a while now using Dafang Hacks.

ultrawyde, 2023-05-16:
Having the same issue. Flashing back to the official firmware works and I am able to get it to join my network.
I get the solid blue light when flashing openmiko, then it goes to solid yellow and finally flashing yellow. But it never joins the network. The few attempts I've made to install openmiko keep corrupting my sd cards as well, and full formats take forever.

ultrawyde, 2023-05-16:
I tried using v0.0.40 and I _finally_ got some logs in the var folder. 
wpa_supplicant.log shows this:
`Line 8: invalid number ""0""
Line 8: failed to parse scan_ssid '"0"'.
Line 14: failed to parse network block.`

The extra quotes around '0' got me to check some documentation on wpa_supplicant.
The wpa_gen.sh sets the scan_ssid equal to "0". Documentation however, states this value should not be surrounded by quotes.
I removed the quotes and now my camera is on the network, albeit only on v0.0.40.

martinbutt, 2023-05-22:
Thanks for the follow up. There is a [PR](https://github.com/openmiko/openmiko/pull/112) to fix this.