openmiko/openmiko #99: Video Drops/Cam Freeze

EQNish, 2023-01-01:
Been using OpenMiko for almost a year on about 12 cams, I like the project, love the RTSP and no Cloud feature, but my Cams Lock/freeze and/or stop sending video.  In order to fix I have to reboot/remove power, any ideas what to tweak to fix this? where should I start?

nosoop, 2023-02-10:
If this is happening every 15 minutes, it's likely because of `wpa_supplicant` being `SIGHUP`'d due to `logrotate`.

My workaround was to create an overlay with the following changes applied:

```diff
--- /etc/init.d/S35wifi
+++ /etc/init.d/S35wifi
@@ -31,7 +31,7 @@
        # https://stackoverflow.com/questions/1251999/how-can-i-replace-a-newline-n-using-sed
        sed -i ':a;N;$!ba;s/\r//g' "$CONFIGPATH/wpa_supplicant.conf"

-       wpa_supplicant -Dnl80211 -i wlan0 -c "$CONFIGPATH/wpa_supplicant.conf" -B -d -f /var/log/wpa_supplicant.log
+       wpa_supplicant -Dnl80211 -i wlan0 -c "$CONFIGPATH/wpa_supplicant.conf" -B -d

        HOSTNAME=`cat /etc/hostname`

--- /etc/logrotate.conf
+++ /etc/logrotate.conf
@@ -24,7 +24,7 @@
     rotate 2
     size 10k
     missingok
-    postrotate
-            /usr/bin/killall -SIGHUP wpa_supplicant
-    endscript
 }
```
You may need to set something up to ensure `/etc/logrotate.conf` has permissions set to `u=rw,go=r`; run `/usr/sbin/logrotate /etc/logrotate.conf` and make sure it's not raising any errors.

martinbutt, 2023-02-20:
This could be the root cause of the issue that is causing the cameras to stop working suddenly when the filesystem becomes corrupted. I've created some branches to remove logging completely [here](https://github.com/martinbutt/openmiko/tree/disable-logging-option) and [here](https://github.com/martinbutt/ingenic_videocap/tree/disable-logging-option). As my PRs are not being merged in this project, I decided to merge together all of my changes from the last six months and [release them for testing](https://github.com/martinbutt/openmiko/releases/tag/v1.0.0-alpha.1). This should address the issue that it stops working after a while. Testing is appreciated, and you can [file an issue](https://github.com/martinbutt/openmiko/issues) in that repo if you have any problems.

bill-mcgonigle, 2023-04-10:
> my Cams Lock/freeze and/or stop sending video

@EQNish also see if your v4l2rtspserver is crashing (can you ssh in?).  I use a script to check mine and restart it.  Not sure why it dies - RAM on my camera might be imperfect (vi isn't even stable, though nano is, for instance).  Cheap computers. :)

KentBrockman, 2023-05-02:
I'm getting this issue as well (sometimes a few others) "this issue" being the RTSP stream dies after sometime. Seems I can't even get it started up:

```
# /etc/init.d/S60camera restart
killall: v4l2rtspserver: no process killed
killall: ffmpeg: no process killed
Adding necessary camera modules
root: Determined image sensor model as jxf23
insmod: can't insert '/driver/sensor_jxf23.ko': File exists
c
c
c
c
c
Audio is set to disabled in openmiko.conf. Skipping starting ffmpeg
Setting up 4 v4l2loopback device(s)
root: Starting video frame capture process
root: Starting v4l2rtspserver
VIDIOC_REQBUFS: Inappropriate ioctl for device
VIDIOC_STREAMOFF: Inappropriate ioctl for device
VIDIOC_REQBUFS: Inappropriate ioctl for device
VIDIOC_REQBUFS: Inappropriate ioctl for device
VIDIOC_STREAMOFF: Inappropriate ioctl for device
VIDIOC_REQBUFS: Inappropriate ioctl for device
VIDIOC_STREAMOFF: Bad file descriptor
VIDIOC_REQBUFS: Bad file descriptor
terminate called after throwing an instance of 'std::bad_alloc'
  what():  std::bad_alloc
```

This is on v0.0.40. MJPEG seems to work fine however. Rebooting the camera sometimes brings things back up.