openmiko/openmiko #102: Never receiving blue light when new firmware is uploaded

WestKleinman, 2023-02-03:
I followed the steps through plugging in the flash drive, and when I do so I get a solid Yellow light that when released after about 30 seconds starts flashing, but never a blue light. This is with the setup button pressed. I have tried following along with the steps after that to see if it is just a fluke but nothing connects and I get no log files. I have already tried multiple SD cards, ranging from 128MB to 64 GB, formatting them all to fat32. Anything else I can try?

jaws35smith, 2023-02-03:
Are you including the wpa and config file?

WestKleinman, 2023-02-03:
can you provide links to those? I thought I just had to first put the demo.bin file on the SD Card, I hadn't gotten to the part where I create the /config/overlay/etc directory

jaws35smith, 2023-02-06:
[https://github.com/openmiko/openmiko/blob/master/overlay_minimal/etc/openmiko.conf](url) config file you may need to adjust if you have a pan vs a v2

[https://github.com/openmiko/openmiko/blob/master/overlay_minimal/etc/wpa_supplicant.conf](url) wpa_supplicant.conf you will have to add your information. 

Once the demo.bin is succesful you should see a var file on the sd card with a wpa_supplicant.conf log. You can look in it for the new MAC address and create a static IP on your router to help reach the new URLs.

bencoleabbott, 2024-04-22:
I know this is really old, but for someone trying this in the future, you have to have an SD card that is 32GB or smaller and formatted with FAT32.