openmiko/openmiko #104: Is There a Setting to Disable IR Lights in Night Mode?

rbassett3, 2023-02-11:
Thank you for this excellent project!

I'm using a WyzeV2 cam against a window, and when the camera switches to night mode it activates the camera's infrared light. I have the camera up against a window, so the red light creates a reflection so I'd like to disable it. I checked through the documentation and settings.json file but didn't see anything relevant.

Is there a way to disable the camera's infrared lights in night mode?

DannySauval, 2023-02-18:
Hey, the IR lights are required for night mode to work, so you would need to disable the night mode. ENABLE_AUTONIGHT=0.

You could also enable the API, and curl -d value=0 http://_CAMERA_IP_:8081/api/ir_led to do what you asked, but I suppose you will end up with a black image when in night mode.

Grntrenchman, 2023-05-31:
> Is there a way to disable the camera's infrared lights in night mode?

If you're still interested.. 
The nightmode is just a script (/usr/bin/nightmode.sh), if you copy it to the same location in the overlay (/config/overlay/usr/bin/), or on the sdcard, you can then edit it to remove the portions that turn the lights on/off (they are separate from the ir-cut), and it will stay persistent as long as it stays in the overlay directories.

If you have SSH access, you can use cp to copy, and nano to edit.
otherwise, you can pull the file from [Here](https://github.com/openmiko/openmiko/blob/master/overlay_minimal/usr/bin/nightmode.sh).

Edit as follows:
```
case $1 in
        on)
                echo "$(date) - nightmode on"
                if [ "$DISABLE_LEDS" == "0" ]; then
                        ir_led on <--- REMOVE THIS LINE
                        ir_cut off
                fi
                echo "1" > $NIGHTVISION_FILE
                ;;
        off)
                echo "$(date) - nightmode off"
                ir_led off <--- REMOVE THIS LINE
                ir_cut on
                echo "0" > $NIGHTVISION_FILE
                ;;

```