openmiko/openmiko #106: is this project still under development ?

carefulcomputer, 2023-02-16:
apologies for noob question, Is this project still being developed ? i wanted to play with this firmware (specially that it doesn't require SD card) . I don't see any recent updates in codebase and when I load the latest version, i can only load the rtsp stream which stops working after some time. From browser when I get to http page of cam, it is blank.

DannySauval, 2023-02-18:
Hey, I know some people are working on the project, it is just not as active as it used to.

For you issues, I had the same when using the latest versions. I downgraded to 0.0.40 and it has been working great for the past few weeks.

carefulcomputer, 2023-02-18:
Thanks. I downgraded to .40 but for some reason it is still not working. Can you please share instructions on how to downgrade (preferably with a complete wipe instructions).

martinbutt, 2023-02-20:
I decided to merge together all of my changes from the last six months and [release them for testing](https://github.com/martinbutt/openmiko/releases/tag/v1.0.0-alpha.1). This should address the issue that it stops working after a while. Testing is appreciated, and you can [file an issue](https://github.com/martinbutt/openmiko/issues) in that repo if you have any problems.

carefulcomputer, 2023-02-20:
@martinbutt , is this IP address expected to be hardcoded here ?  
https://github.com/martinbutt/openmiko/blob/936ec91747e4d32addfd6dcb9b95a57d3a17e61e/web/openmiko/src/store.js#L17

martinbutt, 2023-02-20:
I assume not, but it's not part of [my changes](https://github.com/openmiko/openmiko/compare/master...martinbutt:openmiko:master).

I haven't done any work on the frontend, and assume it's not currently working. I'm happy to code review and test an changes though.

martinbutt, 2023-03-24:
That works for me. I've closed all of my individual PRs and created a [new combined one](https://github.com/openmiko/openmiko/pull/109).

Feel free to create a feature branch instead of merging to master.

carefulcomputer, 2023-03-26:
Thanks to both of you ! I will have to try new one. Quick question, is UI fixed or still broken ?

carefulcomputer, 2023-04-07:
isn't there a web UI (just like https://github.com/EliasKotlyar/Xiaomi-Dafang-Hacks ) to configure the camera and connection settings ?

martinbutt, 2023-04-10:
It looks like @tachang started to build a Vue.js based system [here](https://github.com/openmiko/openmiko/tree/master/web/openmiko) a couple of years ago, but it's not complete.