openmiko/openmiko #108: Which of the latest OM versions work with OctoPrint?

ghnu, 2023-03-03:
Hi, while my ancient demo.bin dated March 27, 2020 worked (not stable, though) with my Wyze Cam v2, as I could use it as web cam for my Windows PC (my function test :slightly_smiling_face: ) newer versions of the OM firmware that I install on my cam - such as 0.0.30 or 0.0.40 - no longer bring them up as webcam on my Windows PC. As a result, Device Manager does not show my flashed Wyze Cam v2 :disappointed:

Instead, the orange led on the cam keeps blinking. What did I do wrong?
I don't want to assign my cam an IP address or bring it into my WLAN at all. My only usage is as webcam (for testing) and for Octoprint on an RPI4.

Any help?

martinbutt, 2023-03-05:
`0.0.40` is the last stable version. It's possible a change in the config file format since you last upgraded is preventing the cameras from working. To start over:
- Hold the reset button for a few seconds and it will flash blue three times to show it has deleted any residual files.
- Reformat the memory card, as it can easily become corrupted if it is writing to a log file when it is powered off.
- Flash the camera with the [0.0.40 firmware](https://github.com/openmiko/openmiko/releases/tag/v0.0.40).
- Use the `0.0.40` [config files](https://github.com/openmiko/openmiko/tree/v0.0.40/overlay_minimal/etc) as a starting point to reconfigure the camera.

Grntrenchman, 2023-05-31:
> Hi, while my ancient demo.bin dated March 27, 2020 worked (not stable, though) with my Wyze Cam v2

Sorry to inform you, but you're probably confusing the official [Wyze Cam v2 Webcam Firmware](https://wyze-firmware.s3-us-west-2.amazonaws.com/V2Webcam.zip) with OM. It's dated 3/27/2020 11:23a (so you can confirm).

There's nothing I can find on the project pages about webcam use or mounting in OpenMiko. Not saying it's not possible, just def not advertised or even documented that I can see.