openmiko/openmiko #111: Camera switching between Day and Night mode constantly in low light.

jamespet77, 2023-04-06:
**Describe the bug**
During failing light.  Ex not quite dark, not quite light. The camera constantly switches from night to day mode/
Previous versions had the issue of switching to night mode after a few hours and not being able to switch back to day mode. This is a new error and the camera works better during dark and light scenarios, but the failing light seems to still be an issue. 

**Specs**
Wyze V2 camera latest openmiko firmware / Same result using current martinbutt firmware

**To Reproduce**
Steps to reproduce the behavior:
1) Turn on camera.
2) fade lights or wait till sunset/sunrise
**Expected behavior**
I would expect that the lights would remain at the last stable state. 
1) If night moving to day then the camera should remain on night mode until it cam clearly visualize a picture in day mode.  
2) If day moving to night then as soon as the camera can not visualize a clear day mode switch to night mode and remain until a clear day mode cam be produced.

Grntrenchman, 2023-05-31:
> The camera constantly switches from night to day mode

I was having this problem in the quasi-dark of the camera location.

I dug around for the settings. In openmiko.conf, you have the "AUTONIGHT_PARAMS".

This is what controls when the trigger is. You can adjust these values for better detection in your specific case.

Since the autonight seems to have come from dafang, while poking around, it referred me to a webpage with configuration that would have been supplied in the included webUI. I managed to find an image of this page in the dafang github, and used the info to adjust my settings accordingly.

For "exposure", lower number means brighter.

Here is the image:
[image]

doctorevil30564, 2025-07-05:
for the example you are showing, what would the entry be in the openmiko.conf file for AUTONIGHT_PARMS? I may have an older version from earlier instructions on setting openmiko up, as mine doesn't even have that section in it.