openmiko/openmiko #119: Crontab overwritten on reboot

darrencope, 2023-06-12:
**Describe the bug**
For some reason, my camera (which I have connected to Synology Surveillance Station via RTSP) loses connection to Synology sporadically, despite it being 'up' everywhere else. A reboot seems to fix the issue. Given this, I want to schedule a reboot via crontab. However, when I add an entry via crontab -e it works the first time, but after the reboot, my entry is no longer present in crontab. It appears to be overwritten somewhere upon reboot, but I can't find where.

**Specs**
Using a WyzeCam v2 with OpenMiko

**To Reproduce**
Steps to reproduce the behavior:
crontab -e and add an entry.
Save.
crontab behaves as expected
reboot
crontab -e - entry is no longer present

**Expected behavior**
crontab entry would persist after reboot

valantur, 2023-06-14:
have you tried adding the crontab file to the etc folder? just like the other config files that can be made permanent as described in the readme file of this project.

darrencope, 2023-06-19:
I have not. I don't know which file is overwriting the crontab, or where I should place it in the etc folder. How would I determine this?

darrencope, 2023-06-20:
I was able to sort this out.

I created the folder /sdcard//config/overlay/var/spool/cron/crontabs/ and then created the 'root' crontab file using the existing crontab (crontab -l > root). By modifying this file, the changes are then written to /var/spool/cron/crontabs/root successfully after reboot.

closed 2023-06-20 by darrencope