openmiko/openmiko #122: openmiko seems to trash the file system of every microsd card I use

flyize, 2023-07-11:
As soon as I try to flash it and then put it back into my computer, the partition is there but it has no file system. I've tried with multiple micrsd cards and multiple cameras. Suggestions?

doucettom, 2023-07-28:
I happen to have the same problem on 3 different cameras.

doucettom, 2023-07-28:
Added informations: Seems like it is creating swap partiton on SD card instead of somewhere else on the device (if a swap is needed at all)

doucettom, 2023-08-01:
OK so I found the source. Problem is with the conf generation scrip that output an incomplete file somehow and it defaults to use a swap partition on the SD card (that it seemingly creates) Re-doing a file manually brought back up my WCv2 :) I just cannot explain why that configuration stayed there even after reflashing wyze firmware though.

GasMeister, 2023-10-29:
I think I am having the same error.

Upon booting the camera with a freshly formatted SDcard, it successfully mounts the partition:
`# df -h
Filesystem                Size      Used Available Use% Mounted on
devtmpfs                 44.6M         0     44.6M   0% /dev
/dev/mmcblk0p1           29.7G     96.0K     29.7G   0% /sdcard
/dev/zram0               46.5M     32.9M     10.2M  76% /
tmpfs                    44.8M      8.0K     44.8M   0% /dev/shm
tmpfs                    44.8M     56.0K     44.8M   0% /tmp
tmpfs                    44.8M     28.0K     44.8M   0% /run
/dev/mmcblk0p1           29.7G     96.0K     29.7G   0% /sdcard
/dev/mtdblock3          256.0K    132.0K    124.0K  52% /config`

`# mount
rootfs on / type rootfs (rw)
devtmpfs on /dev type devtmpfs (rw,relatime,size=45680k,nr_inodes=11420,mode=755)
proc on /proc type proc (rw,relatime)
sysfs on /sys type sysfs (rw,relatime)
/dev/mmcblk0p1 on /sdcard type vfat (rw,relatime,fmask=0000,dmask=0000,allow_utime=0022,codepage=437,iocharset=iso8859-1,shortname=mixed,errors=remount-ro)
/dev/zram0 on / type ext2 (rw,relatime)
devpts on /dev/pts type devpts (rw,relatime,gid=5,mode=620,ptmxmode=000)
tmpfs on /dev/shm type tmpfs (rw,relatime,mode=777)
tmpfs on /tmp type tmpfs (rw,relatime)
tmpfs on /run type tmpfs (rw,nosuid,nodev,relatime,mode=755)
/dev/mmcblk0p1 on /sdcard type vfat (rw,relatime,fmask=0000,dmask=0000,allow_utime=0022,codepage=437,iocharset=iso8859-1,shortname=mixed,errors=remount-ro)
/dev/mtdblock3 on /config type jffs2 (rw,relatime)`

The cards seems to work fine until I pull it.  Windows always wants to reformat the device.  Also, rebooting the camera causes the image to fail mounting.  I looked through the conf generation script and could not find the culprit.  

@doucettom - Can you tell us what file you brought in manually?

GasMeister, 2023-10-29:
**_FACEPALM_**

I read through RC scripts and noticed the S21swap script.  It was reading /etc/openmiko.conf to determine where to create swap, which was the first partition on the card (over the FAT partition I created on Windows).  I edited openmiko.conf and set ENABLE_SWAP=0.  I also set SD_PARTITION=/dev/mmcblk0p1.  

Now everything operates normally.