openmiko/openmiko #131: Blank Page with Bootstrap 101 Template

ITCMD, 2024-01-08:
Cant seem to get this to work on my Wyze Cam V2.

Formatted an SD Card as fat32, put v40 named demo.bin in there, put in wpa supplicant and the auto generated config files. It does connect to my network, but rtsp connections time out and when I connect with any URL via HTTP it is a blank screen with the title "Bootstrap 101 Template"

Any tips?

Also tried the config file from #55 because all I really want is an RTSP stream, but same issue.

martinbutt, 2024-01-08:
"Bootstrap 101 Template" is what you see on a running camera if you just hit it on port 80. If you have the MJPEG over HTTP stream enabled in your config (`ENABLE_MJPEG_OVER_HTTP=1`), then you should see that on http://Your_Camera_IP:8080/?action=stream

As for the RTSP streams, make sure you have the `/etc/v4l2rtspserver.conf` file set up correctly.

Also, make sure your files have UNIX-style line endings. This is a older bug that still remains in v0.0.40, but has been fixed in the v1.0.0.alpha releases.

If you still have no luck, share your config files here.

ITCMD, 2024-01-09:
In the config given in #55 those files were not present.
I'm looking to create a dead simple RTSP server with a username and password and automatic night vision to use with ZoneMinder (which is a nightmare to config in itself).

Any chance you could provide a config setup for that? v1 or v0.40 are both fine.

ITCMD, 2024-01-09:
[config.zip](https://github.com/openmiko/openmiko/files/13867558/config.zip)

For reference, all I have are these config files (with wpa_supplicant updated for my network)

martinbutt, 2024-01-10:
I've adapted my config, which is for `v1.0.0.alpha`, to work with `v0.0.40`, but I don't have a camera with `v0.0.40` to test it.

[config-0.0.40-zoneminder.zip](https://github.com/openmiko/openmiko/files/13883057/config-0.0.40-zoneminder.zip)

I'd remove any other config files that where there before, get this working, then try adding them back after. You'll need to edit the provided `v4l2rtspserver.conf` and add a username and password. Once you have done this, you should be able to test the RTSP feed using VLC on `rtsp://YOUR_USERNAME:YOUR_PASSWORD@your_camera_ip:8554/video3_unicast`.

I have two streams set up. One on RTSP at 25 fps, and one at 5 fps using MJPEG. My ZoneMinder is configured to monitor the 5 fps feed. If you try to monitor too many high res or high fps streams in ZoneMinder, you'll quickly overwhelm the CPU, so it's best to only send the frames you need for the analysis. 

In ZoneMinder configure two cameras. One for monitoring and one for recording. In the settings for the camera 1, you'll need to set General > Function to `modect`, and Source > Source Path to `http://your_camera_ip:8080/?action=stream`. In the settings for camera 2 set General > Function to `nodect`, General > Linked Monitors select camera 1, and Source > Source Path to `rtsp://YOUR_USERNAME:YOUR_PASSWORD@your_camera_ip:8554/video3_unicast`. Now when camera 1 goes into an alarm state, it will trigger recording on camera 2.