openmiko/openmiko #135: pan tilt wyze cam pan v2

openmotion, 2024-02-21:
Hello
i cannot find the command to change pan and title for the cam pan v2
so is it possible to add this command ? ( binary or script )

Yujkore, 2024-04-01:
I have the WyzeCam Pan V3. I haven't gotten ballsy enough to see if openmiko even worked on that one, but I was also curious how the firmware would address the pan/tilt. I am also curious about this feature.

kult, 2025-01-06:
I just got Openmiko running on a WYZECP1. I also hope pan/tilt controls are (or will be) possible.

martinbutt, 2025-01-06:
If you want to play with this, it's likely controlled by a GPIO pin, in the same way the LEDs are. [This code turns the IR LED on and off](https://github.com/openmiko/openmiko/blob/00f72a6a2f701a038b39949fce239d54cd221065/overlay_minimal/usr/bin/nightmode.sh.).

You can do something similar and try setting the various GPIO pins to high (1) and low (0) and see if any of the motors start. You can find [more info on LIBGPIO here](https://github.com/Ventto/libgpio).