openmiko/openmiko #137: NTP

Yujkore, 2024-04-01:
Love the firmware, but for the life of me, I cannot figure out to change the timezone.

I have a local NTP server that I use, as I don't want my WyzeCam to report back to the mother ship--or any other remote location for that matter. I tried creating the file, /etc/TZ with "America/Los_Angeles", but openmiko still uses UTC, which is 5 hours off.

Please assist and thanks in advance!

kontiainen, 2024-04-30:
Having the same problem. TZ file gets copied into /etc and has "Europe/Helsinki" in it, but camera shows the UTC time. Ideas?

Yujkore, 2024-05-01:
I forgot whet i found the answer, 
but i set the file,  config/overlay/etc/TZ  to "PST8PDT,M3.2.0,M11.1.0". I don't know what any of that cap means, but it fixed me.

kontiainen, 2024-05-01:
Ok, I will try the ”EET-2EEST,M3.5.0/3,M10.5.0/4” string. Let’s see.

ndrwstn, 2024-09-01:
It's called POSIX time zone formats. For other people who are having problems, here's link to some documentation: [POSIX and Olson time zone formats](https://developer.ibm.com/articles/au-aix-posix/).

I'd just ask Claude or Chat GPT to generate them for you based on your location, it will most likely get it right.

ufocia, 2026-09-23:
Time zone definitions are saved in /usr/share/timezones.tsv. Use grep with > to create your /config/overlay/etc/TZ .